justcallmekoko/ESP32Marauder #2: Compiler error:  'class TFT_eSPI_Button' has no member named 'drawButton2'

tobozo, 2019-11-19:
the best I could come up with to end this error was to add this function to TFT_eSPI, not sure what yours is doing though:

```C
void TFT_eSPI_Button::drawButton(String label, boolean inverted) {
  strncpy(_label, label.c_str(), 9);
  drawButton( inverted );
}
```

looks like it can't be wrapped outside TFT_eSPI anyway

tobozo, 2019-11-19:
kinda working with a LoLin D32 Pro although the Touch doesn't seem to work, I may have to double-check the wiring, thanks for this great project !!

[image]

justcallmekoko, 2019-11-19:
@tobozo Thank you for reminding me. I did extend the TFT_eSPI library by adding the drawButton2 function. I can't remember off the top of my head what I have it doing but I will let you know as soon as I sit down at my project PC.

To verify your wiring, check the [schematic](https://github.com/justcallmekoko/ESP32Marauder/blob/master/schematics/Schematic_ESP32-Marauder-2_ESP32-Marauder-2-Schematic_20191007113616_png.png) I included in the repo. The connections on the marauder may be different from the connections you made on your lolin but the inter-pin connections on the TFT will most likely be the issue. Just check and see which pins are part of the same Net on my schematic.

Also check to make sure the tft screen has the touch controller chip on the back. I know some manufacturers ship these screens without the touch controller.

PS. Great job getting it working. Thanks for sharing the picture and the progress.

justcallmekoko, 2019-11-19:
@tobozo 
Added this line to [line 22] (https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Button.h#L22) of Button.h in the Extensions of Bodmer's TFT_eSPI library
```C
void     drawButton2( String button_name = "", boolean inverted = false);
```

and this is my `drawButton2` function in Button.cpp:
```C
void TFT_eSPI_Button::drawButton2(String button_name, boolean inverted) {
  uint16_t fill, outline, text;

  if(!inverted) {
    fill    = _fillcolor;
    outline = _outlinecolor;
    text    = _textcolor;
  } else {
    fill    = _textcolor;
    outline = _outlinecolor;
    text    = _fillcolor;
  }

  uint8_t r = min(_w, _h) / 4; // Corner radius
  _gfx->fillRoundRect(_x1, _y1, _w, _h, r, fill);
  _gfx->drawRoundRect(_x1, _y1, _w, _h, r, outline);

  _gfx->setTextColor(text);
  _gfx->setTextSize(_textsize);

  uint8_t tempdatum = _gfx->getTextDatum();
  _gfx->setTextDatum(ML_DATUM);
  if (button_name == "")
	_gfx->drawString(_label, _x1 + 20, _y1 + (_h/2));
  else
	_gfx->drawString(button_name, _x1 + 20, _y1 + (_h/2));
  _gfx->setTextDatum(tempdatum);
}
```

closed 2019-11-19 by justcallmekoko

tobozo, 2019-11-19:
@justcallmekoko thanks! turns out T_CS was wired to the wrong pin :man_facepalming: so touch support is now tested and working as a charm :tada:

lukorely, 2020-06-01:
when trying to upload the code I skip this in menufunctions.cpp
'class TFT_eSPI' has no member named 'setTouch'

roschme, 2021-02-22:
when trying to upload the code I skip this in menufunctions.cpp
'class TFT_eSPI' has no member named 'setTouch'

Same here unfortunately...

roschme, 2021-02-22:
ok, you'll have to config Bodmer's TFT_eSPI library take a look here: 
https://github.com/Bodmer/TFT_eSPI/issues/454#issuecomment-548085768

roschme, 2021-02-22:
Display.h
Line 50 ->> #define TFT_DIY

roschme, 2021-02-22:
thx@koko great work!

RChadwick7, 2021-02-27:
I have this same issue. I configured Bodmer's library by commenting and uncommenting the proper lines in User_Setup_Select.h, and I uncommented line 50 in Display.h. Is there anything else I'm missing?

roschme, 2021-02-28:
> 
> 
> I have this same issue. I configured Bodmer's library by commenting and uncommenting the proper lines in
User_Setup_Select.h, and I uncommented line 50 in Display.h. Is there anything else I'm missing?

If you use the ILI9341, uncomment this in User_Setup_Select.h : 

(Line 22)

#define ILI9341_DRIVER      

AND this (line 162 and following; Pins according to "koko's" wiring diagram)

// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP32 SETUP   ######

// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   17  // Chip select control pin
#define TFT_DC    16  // Data Command control pin
#define TFT_RST   5  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TFT_BL   32  // LED back-light (only for ST7789 with backlight control pin)

#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

//#define TFT_WR 22    // Write strobe for modified Raspberry Pi TFT only

RChadwick7, 2021-03-01:
That fixed the issue, thanks!

tigrik191, 2021-03-10:
a lot of library errors, give someone realties all the necessary libraries for this project in the archive

Arduino: 1.8.13 (Windows 10), Плата:"ESP32 Wrover Module, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), QIO, 80MHz, 921600, None"

Несколько библиотек найдено для "WiFi.h"

In file included from c:\users\dmitriy\documents\arduino\libraries\lvgl\src/lv_misc/lv_log.h:16:0,

 Используется: C:\Users\Dmitriy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\WiFi

                 from c:\users\dmitriy\documents\arduino\libraries\lvgl\lvgl.h:25,

Не используется: C:\Program Files\libraries\WiFi

                 from C:\Users\Dmitriy\Documents\Arduino\libraries\lvgl\src/lvgl.h:17,

Несколько библиотек найдено для "SD.h"

                 from sketch\Display.h:11,

 Используется: C:\Users\Dmitriy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\SD

Не используется: C:\Program Files\libraries\SD

                 from C:\Users\Dmitriy\Desktop\ESP32Marauder-master\esp32_marauder\esp32_marauder.ino:20:

c:\users\dmitriy\documents\arduino\libraries\lvgl\src\lv_conf_internal.h:40:101: fatal error: ../../lv_conf.h: No such file or directory

compilation terminated.

exit status 1

Ошибка компиляции для платы ESP32 Wrover Module.

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"