justcallmekoko/ESP32Marauder #6: Is correcting the compiler error still a part of the flashing process?

Ahorrorfilm, 2020-01-30:
https://github.com/justcallmekoko/ESP32Marauder/issues/2#issuecomment-555695918

justcallmekoko, 2020-01-30:
It isn't an error with the compiler. It is an extension of the TFT_eSPI library that has to be made in order to get the buttons lined up properly. But the referenced issue is still part of the process

Ahorrorfilm, 2020-01-30:
Okay gotcha, thanks! Does that need to be done with a text editor before its added as a library? Having trouble editing after it's already added. Sorry, this is my first time messing with arduino/esp32 stuff..

justcallmekoko, 2020-01-30:
It can be done before or after. It should not matter. 
If you want to do it before, then you can.
If you want to wait until you have installed the library and just traverse the Arduino directories and modify it there.

Ahorrorfilm, 2020-01-30:
okay word, thanks for being chill also.

justcallmekoko, 2020-01-30:
No problem. Let me know if you have any other questions.

closed 2020-01-30 by justcallmekoko