justcallmekoko/ESP32Marauder #9: M5Stack implementation roadmap

tobozo, 2020-02-05:
As [per request](https://github.com/justcallmekoko/ESP32Marauder/issues/5#issuecomment-581161878) I'll leave here the steps needed to get the ESP32Marauder to work on the M5Stack.

  - Replace TFT_eSPI by ESP32-Chimera-Core (also supersedes TouchButton and JPEGDecoder)
  - Create HID strategy (must be agnostic on the HID source)
  - Rewrite the existing Touch support in a pluggable format
  - Disable touch support (won't be loaded by TFT_eSPI)
  - Enable button support in a pluggable format

Random observations:

  - Some HID event detection is still handled from within WiFiScan.cpp, should be moved to MenuFunction
  - Enabling all compiler warnings makes the build fail, Display.cpp probably needs a complete rewrite
[image]
  - code still contains `#ifdef ESP8266` statements and will be cleaned up, this can't end up with a clean PR
  - Free_Fonts.h has too many different versions going around for not being included in this project
  - PROGMEM keyword is ignored on ESP32, it is redundant with `const`
  - None of the WiFiScan.cpp callbacks should ever draw or read touch => will probably need to be rewritten to use asynchroneous signals or mutexes

justcallmekoko, 2020-02-05:
Created the gitter. Closing this issue to continue there.

closed 2020-02-05 by justcallmekoko

elixx, 2020-06-18:
I am interested in the M5Stack implemenation -- Any more notes or development on this?

tobozo, 2020-06-18:
@elixx just a proof of concept with partial button support so far, at the time of writing the code, implementing buttons was done by partially removing the touch logic and replacing TFT_eSPI by ESP32-Chimera-Core

A complete UI rewrite is probably necessary to get all touch functionalities triggered by buttons, and I'm not sure the M5Stack has enough of them.

TTGO-Twatch is probably a better "other" platform for this project, will get one soon in the mailbox and post feedback here :-)

elixx, 2020-06-18:
Yeah, I had similar thoughts about the lack of buttons, but was happy to see someone tread similarly, already. 
TTGO-Twatch looks fancy!

Gotcha007, 2021-03-03:
Does that mean that the new one M5Stack Core2 will work better. They are saying it has a capacitive touch screen

pramady101, 2021-08-16:
Any further update on this? I'm hoping I can run maurader on a core 2?