justcallmekoko/ESP32Marauder #10: Update Firmware issue

AvgJxmes, 2020-02-22:
While connected to SSID MarauderOTA via 192.168.4.1, whenever I try to update the firmware, i keep getting Not found:/serverIndex.

[image]

AvgJxmes, 2020-02-22:
The reason why it wasn't loading was because of this <script src='https://ajax.googleapis.com/ajax/libs/jquery/3.2.1/jquery.min.js'>

hence need to used ethernet cable to connect to the internet for the computer and then wireless join to the marauder.

AvgJxmes, 2020-02-22:
Pls update to the latest release of the Marauder firmware currently the bin file is still 4.0. If possible can update the entire repo to the latest version?

tobozo, 2020-02-22:
have you tried to edit [Web.cpp](https://github.com/justcallmekoko/ESP32Marauder/blob/eb492c4cb7f9a7a26a2816c24448075ca8897d1e/esp32_marauder/Web.cpp#L47) and replace `WiFi.softAP(ssid, password);` by this :

    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);

you may need to remove these lines too:

    Serial.print("IP address: ");
    Serial.println(WiFi.softAPIP());

Explanation: instead of being an access point, your Marauder will join your WAN and be accessible via mdns (e.g. esp32.local) or IP address and will share the internet connection.

Other solution would be to bundle jquery.min.js with the SPIFFS partition and modify the HTML, but there's probably a reason why @justcallmekoko hasn't done it already (browser cache problems or the need to hardcode your gateway ssid/pass ?).

justcallmekoko, 2020-02-22:
Not sure why, but i did not have an issue running the update from my devices while marauder was running it’s own AP. I will probably apply the solution shown in [this thread](https://esp32.com/viewtopic.php?t=11744) on the espressif forum.

tobozo, 2020-02-22:
interesting to note that ``server.serveStatic("/js", SPIFFS, "/js"); will automatically generate the Cache-Control headers, decide whether to send the .js or .js.gz file depending on the request, and prevent the browser to mess with the stability of the webserver

justcallmekoko, 2020-02-22:
I believe I have resolved the issue thanks to the thread I referenced earlier. Please refer to the following screenshots. During this testing, the marauder was hosting its own access point with no internet access. I will publish a new release (v0.4.5)

**Here is the login page of the update server**
[image]

**Here is the browse file page with an asyncwebserver warning**
[image]

**File has been chosen (without reloaded the page)**
[image]

**Starting the update**
[image]

**jquery function called**
[image]

**Update running from perspective of the marauder UI**
[image]

**Update applied (with older version of marauder)**
[image]

AvgJxmes, 2020-02-23:
> have you tried to edit Web.cpp and replace WiFi.softAP(ssid, password); by this :
> WiFi.mode(WIFI_STA);
> WiFi.begin(ssid, password);
>
> you may need to remove these lines too:
> Serial.print("IP address: [...]

done thanks.

AvgJxmes, 2020-02-23:
> I believe I have resolved the issue thanks to the thread I referenced earlier. Please refer to the following screenshots. During this testing, the marauder was hosting its own access point with no [...]

got it thanks for the quick reply and help. Enjoying the ESP32 Marauder now. A lot of people from office asking me where I got it.

closed 2020-02-23 by AvgJxmes