justcallmekoko/ESP32Marauder #25: Support for ST7789V

RChadwick7, 2020-06-18:
The new TTGO watches look like an ideal platform for the ESP32Marauder.

RChadwick7, 2020-06-25:
I originally got my watch on Tindie https://www.tindie.com/products/ttgo/lilygor-ttgo-t-watch-2020/
They sold out a few times, so it might not be available there. I also recently saw it on Ebay for a few dollars more:
https://www.ebay.com/itm/LILYGO-TTGO-T-Watch-2020-ESP32-Main-Chip-1-54-Inch-Touch-Display-Programmable/153973300581

justcallmekoko, 2020-06-25:
Just placed my order

justcallmekoko, 2020-07-13:
@RChadwick7 Received my TTFO Watch today. Gotta do some work on it to see what it has under the hood.

RChadwick7, 2020-07-14:
I'm very excited to see what can come from this.

tobozo, 2020-07-14:
I'm curious how long the TTGO TWatch 2020 model will last (no SD Card, so using SPIFFS only) before the flash memory wears out.

I've attempted to get this working on the older model, which is bigger but comes with a micro SD card slot, but only succeeded at losing myself in the TFT_eSPI / lvgl configuration :rofl:

RChadwick7, 2020-07-14:
I was a bit disappointed in the lack of SD, but I suppose sacrifices needed to be made.  I hope to poke around soon in mine, hoping the SPI Flash is solderable.

zhaohuansi, 2020-09-19:
> @ RChadwick7今天收到了我的TTFO手表。要对它做一些工作，以查看其内部功能。

how's the progress?

tobozo, 2020-09-19:
@zhaohuansi if you're willing to help on TWatch implementation there's a partial LGFX support on [this fork](https://github.com/tobozo/ESP32Marauder-1/ but you will have to tweak the [display and touch setup](https://github.com/tobozo/ESP32Marauder-1/blob/master/esp32_marauder/esp32_marauder.ino#L75) as document on the [LovyanGFX](https://github.com/lovyan03/LovyanGFX) driver repository and also find a workaround for the power management

justcallmekoko, 2021-06-07:
This will likely need another branch and someone else with the time and resources to make this work.

tobozo, 2021-06-07:
The [hackwatch](https://github.com/virtualabs/hackwatch) seems a better fit for T-Watch+ST7789

tobozo, 2021-06-07:
there are livecoding sessions on @virtualabs's channel https://www.twitch.tv/virtualabs