justcallmekoko/ESP32Marauder #26: Employ the LittleVGL Library

justcallmekoko, 2020-06-19:
@tobozo
I wanted to get your opinion on this before I do it. I came across this embedded graphics library that looks really good. Pretty soon I will start needing on screen keyboards and check boxes for the marauder settings and attack features. Here is a [demo video](https://www.youtube.com/watch?v=dErIWm-IVhc) of the library and I believe it is a 3 part series. You can find the library itself [here](https://github.com/littlevgl/lv_arduino) and a link to the demos used in the videos [here](https://github.com/0015/IdeasNProjects/tree/master/ESP32_LITTLEVGL_ARDUINO_IDE).

Let me know what you think. It seems like it's exactly what I need and it has a small memory footprint.

tobozo, 2020-06-19:
@justcallmekoko LVGL is good, it can easily be added to existing display drivers such as TFT_eSPI or LovyanGFX

The only thing I don't like about this library is the use of snake case, it gives the code a low level feeling that isn't suitable for a presentation layer :)

justcallmekoko, 2020-06-23:
So just posting the update here even though we discussed it on gitter.

The lvgl library takes up about 60k of memory which does not leave enough for running bluetooth functions. While I can always switch to using the esp32 wrover which has an extra 8mb of psram, I would prefer to get this working on the existing hardware considering this is the hardware owned by those who have already purchased a marauder.

What I think I am going to do is make a copy of the marauder firmware and rewrite the entire menu structure to be built around lvgl. I think this would probably have some benefits in the long run because their memory foot print for making a UI is probably a lot smaller than the inefficient UI I created. 

A couple things I am concerned about are the scrolling pages for displaying sniffed devices and the line graphs for the packet monitors. I want to make sure I can still use those features WITH lvgl.

closed 2020-09-07 by justcallmekoko