justcallmekoko/ESP32Marauder #40: Include NimBLE as part of the installation documentation

justcallmekoko, 2020-08-10:
Pretty self explanatory.

Sporksan, 2020-08-15:
Suggestion as to where we can find the library?

justcallmekoko, 2020-08-17:
@Sporksan 
[here](https://github.com/h2zero/NimBLE-Arduino) is the library

Sporksan, 2020-08-17:
Yeah, i should have just googled first but  thanks for pointing at the officially supported version!

justcallmekoko, 2020-08-18:
@Sporksan 
No problem. You can never be too careful with trying to get the correct libraries. Developers can use specific forks of a well known library that can have some minor changes from the original library which make all the difference.

It's on me to include this in the documentation.

justcallmekoko, 2020-09-04:
@tobozo I didn't even think to search for this in the library manager. Good find

justcallmekoko, 2021-04-29:
Added to the wiki

closed 2021-04-29 by justcallmekoko