justcallmekoko/ESP32Marauder #41: v0.6.9 don't work on TTGO T8

klazarev, 2020-08-10:
On screen
[image]

On Serial

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
Using TFT DIY
clearScreen()
SPI_FREQUENCY: 27000000
SPI_READ_FREQUENCY: 20000000
SPI_TOUCH_FREQUENCY: 2500000

--------------------------------

         ESP32 Marauder      

            v0.6.9

       By: justcallmekoko

--------------------------------

RAM Free: 4392020 bytes
SD: SDSC Mounted
SD Card Size: 1886MB
SD Card supported
RAM Free: 4358264 bytes
RAM Free: 4358264 bytes
RAM Free: 4358264 bytes
RAM Free: 4358168 bytes
IP5306 I2C Supported: false
RAM Free: 4358008 bytes
Setting neopixel to black...
initScrollValues()
setupScrollArea()
   tfa: 48
   bfa: 0
yStart: 48
Bulding buttons...
Displaying current menu...
clearScreen()
```

 It compiles fine on Arduino 1.8.13. 
My changes - SPI.begin (14, 2, 15, 13) before SD.begin (SD_CS) in SDInterface.cpp and reassigned SPI pins for display in User_Setup.h

v0.6.8. works great with the same changes

justcallmekoko, 2020-08-11:
So what was the issue?

klazarev, 2020-08-13:
The problem was not solved. On the same IDE with the same changes for TTGO T8 Marauder v0.6.8 compiles and works, and v0.6.9 compiles and does not work.

justcallmekoko, 2020-08-15:
And what happens when you try to install v0.6.9? I couldn't tell from your previous message

justcallmekoko, 2020-10-22:
@klazarev Any updates on the issue you are having. I am still not sure what the problem is.

Also have you tried with v0.8.0?

justcallmekoko, 2020-10-28:
@klazarev Just a heads up. I plan to close this issue next week. If you have any other information, please let me know so we can get your issue resolved.

closed 2020-11-03 by justcallmekoko