justcallmekoko/ESP32Marauder #50: Sorry. This device can attack bluetooth like wifi is not

vovanhau, 2020-08-28:


vovanhau, 2020-08-28:
esp8266_deauther is a very effective wifi attack.
Is bluetooth attack effective

justcallmekoko, 2020-08-28:
@vovanhau 
I am not sure which bluetooth attack you are referring to.
The esp8266 deauthentication is unrelated to the ESP32 since the ESP32 cannot send deauthentication frames using the current esp-idf.

astronomicalunit, 2020-09-01:
The ESP32 marauder is a pretty well rounded tool that one who needs both defensive and offensive capability when out testing new security, conducting a crime, or getting a number 3 with animal style fires and a milk shake at  in n out. It’s a tool you always want with you and I can’t go into details on my use as contracts may not allow me too, because I’m that boring, but it is one of only  7 out of 247 tools that I carry with me every day no matter what. Koko crushed it on this and I’m stoked on the new update. Hoping version 8 can get up in there super deep when getting up in there super deep is required or requested depending if work or with a needy gf so what I mean is I hope version 8.0 has some way to pause and examine stuff in interface as too create beacons etc. But it allows you to scan probe and beacon requests from surrounding area, detect Bluetooth enabled skimmers that unfirtsntky in my state doesn’t permit Me to John wick the ones I find Where as in  mex or tex or Az I could , and it detects pwnagotchis nearby which can give one both advanced notice of attacks nearby, and through sd card one can go back and dig for more info. In WiFi koko is correct it isn’t a Deauther yet it detects deauths through the interface while driving a looking at the pretty colors in packet monitor mode and almost crashing while doing so, as well as seeing other traffic volume per channel in a given area that used with a Deauther can make evil twin attacks for contracted legal jobs, and other features include it looks bad ass and when I see elderly folk I don’t like I point a WiFi pineapple antenna I’ve attached to it at them and tell them I’m controlling their thoughts, plus rick roll beacon and beacon spam to protect or make a WiFi attack against ones network take longer, giving the wise one time to counter and put a water bottle over the barrel of ones broomstick and help them with a full cranial evacuation..If I’m having a good day and if not then a vacation to Mexico to make Kiki tapes sound like Beyoncé or a kids show.. then you can draw which is my favorite thank you koko and you can do much more. But my thumb hurts and all this talk now I’m going to go play with my marauder(s)  excuse errors I am using talk to text as I use my broken phone for regular internet or social negus vs my ones for covert cons or just making my right arm stronger.. it would be cool if some way to add communication or ability it intercept  security cams or if a new model comes out an ability to rfid and add scanner for near by radio etc but for now this is a 10/10 100%

justcallmekoko, 2020-09-01:
@astronomicalunit Messages like these are the reason I enjoy doing what I do. I am happy to hear that this device has some actual applications in the real information security industry. Being the guy who made the device, I lose the appreciation for what this device is actually capable of and I constantly feel like it is missing so much. 

As far as capabilities go, I already have plans to start working with the keyboards some more.  
I plan on adding:
- Sending single handcrafted packets
- Establish WiFi network for some WLAN communication
- Settings
  - Enable/disable PCAP saving
  - Channel switcher

My biggest goal is to be able to add something that would allow for firmware plugins. Like if you want to solder your own rfid or LoRa module to the marauder, you should be able to just add a firmware module to the sd card for marauder to load. I have no idea how to do this or if it is even possible with arduino but if I can make something like that, that would be v1.0.0 material right there.

astronomicalunit, 2020-09-28:
Loving the new update and ability to add SSID! Thank you for the hard work. Have a nice week. Looking forward to future projects and updates..I appreciate the time you’ve taken to help with the screen and answers to a few questions. YouTube video always was a great dive into just how much work goes into them, it well just some of it. Anyways  have a nice day

closed 2020-10-22 by justcallmekoko