justcallmekoko/ESP32Marauder #57: Not an issue, but dsiplay question

Jenpatrick, 2020-10-19:
Just wondering if this is the right display? (for folks to DYI it)

I know between manufacturers overseas, there tends to be huge variances.. but this seems to match the pin outs.. for voltage, spi, etc

https://www.amazon.com/WIshioT-Display-ILI9341-Controller-Arduino/dp/B07916R3KY

Weirdly, I think I have one kicking around in my parts bin from another project.  I'll have to try and report back.  I have like 4 ESp32's kicking around so not a huge deal to try this later this week, and say if that display works.  

Although I wonder if maybe finding a non touchscreen display via i2c, and a clicker, or joystick type control would be a nice alternative.  But GUI work is really not my thing.  Someone would have to redo the input stuff for that i'd assume.

justcallmekoko, 2020-10-19:
@Jenpatrick it looks like that display should work right out of the box with the firmware.
The TFT driver is ILI9341 of course and the touch controller is still an XTP2046.

Jenpatrick, 2020-10-19:
Wonderful.  I will give that a try.  in regards to using an i2c dsiplay, with maybe a menu or joystick dial, any thoughts?

Also I did sign up on Tindie to wait on a v6 of a prod model. (gotta support folks like you!)

 I love this, this just seem so incredibly fun, and neat, and better than the 8266 ones I've seem out there.  Plus I've been mostly using my esp32 stuff for lora sensing and mqtt and The-things-network, for monitoring a remote barn on my property, hardly what I'd call really fun.  :-)  So I like to see stuff like this. versus moisture sensing in a barn, and ambient temperature..  yea, I'ma little boring with my esp32 endeavours.

(Feel free to close this)

astronomicalunit, 2020-10-20:
Hey, koko, my apologies, I’ve been out of country, and have come to realize with over 15 proton emails I forget which one I’m logged into.

So my bad I did not intentionally ignore you. I actually am about to buy one of those arm chats. I too am very interested in Lora. My primary email is opensourcesolutions@proton

I took a break from IG, and well I kind of have too both for sanity and when abroad but I hope you’re having a great week so far. I get notifications on my signal, and the email listed above. For some reason as soon as I close my proton app when logged in as this account it signs me out. So I logged in today and have hundreds of emails and was wondering why the jobs I applied to never reached back. They did unfortunately lol.

Have a nice evening, man!🤜🤛

Sent from ProtonMail Mobile

justcallmekoko, 2020-10-22:
@Jenpatrick It has been a while since I have worked with an I2C display so I would not be able to speak to that at the moment. As far as a joy stick is concerned, I am working on a little something right now and you might like the final product.

Thanks for the support and hopefully I will have some V6s in stock soon.

@astronomicalunit Technology loves to prove just how much of a liability it can be. Hopefully those employers were/are understanding. Good luck, man.

closed 2020-10-22 by justcallmekoko