justcallmekoko/ESP32Marauder #58: Manually compile an error about TFT

LuckyFishGeek, 2020-10-21:
`C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\Display.cpp: In member function 'void Display::RunSetup()':
Display.cpp:42:7: error: 'class TFT_eSPI' has no member named 'setTouch'
   tft.setTouch(calData);
       ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\Display.cpp: In member function 'void Display::drawStylus()':
Display.cpp:482:25: error: 'class TFT_eSPI' has no member named 'getTouch'
   boolean pressed = tft.getTouch(&x, &y);
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\MenuFunctions.cpp: In function 'bool my_touchpad_read(lv_indev_drv_t*, lv_indev_data_t*)':
MenuFunctions.cpp:47:34: error: 'class TFT_eSPI' has no member named 'getTouch'
   bool touched = display_obj.tft.getTouch(&touchX, &touchY, 600);
                                  ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\MenuFunctions.cpp: In member function 'void MenuFunctions::main(uint32_t)':
MenuFunctions.cpp:322:29: error: 'class TFT_eSPI' has no member named 'getTouch'
   pressed = display_obj.tft.getTouch(&t_x, &t_y);
                             ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\MenuFunctions.cpp: In member function 'void MenuFunctions::orientDisplay()':
MenuFunctions.cpp:668:19: error: 'class TFT_eSPI' has no member named 'setTouch'
   display_obj.tft.setTouch(calData);
                   ^
WiFiScan.cpp:1141:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6
         DynamicJsonBuffer jsonBuffer;
           ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunLvJoinWiFi(uint8_t, uint16_t)':
WiFiScan.cpp:439:19: error: 'class TFT_eSPI' has no member named 'setTouch'
   display_obj.tft.setTouch(calData);
                   ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunPacketMonitor(uint8_t, uint16_t)':
WiFiScan.cpp:604:19: error: 'class TFT_eSPI' has no member named 'setTouch'
   display_obj.tft.setTouch(calData);
                   ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunEapolScan(uint8_t, uint16_t)':
WiFiScan.cpp:651:19: error: 'class TFT_eSPI' has no member named 'setTouch'
   display_obj.tft.setTouch(calData);
                   ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\WiFiScan.cpp: In static member function 'static void WiFiScan::pwnSnifferCallback(void*, wifi_promiscuous_pkt_type_t)':
WiFiScan.cpp:1141:27: error: 'jsonBuffer' was not declared in this scope
         DynamicJsonBuffer jsonBuffer;
                           ^
WiFiScan.cpp:1143:19: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6170_91::ObjectRef}' has no member named 'success'
         if (!json.success()) {
                   ^
WiFiScan.cpp:1150:16: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6170_91::ObjectRef}' has no member named 'printTo'
           json.printTo(json_output);
                ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\WiFiScan.cpp: In member function 'void WiFiScan::eapolMonitorMain(uint32_t)':
WiFiScan.cpp:1774:31: error: 'class TFT_eSPI' has no member named 'getTouch'
     pressed = display_obj.tft.getTouch(&t_x, &t_y);
                               ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_223464\sketch\WiFiScan.cpp: In member function 'void WiFiScan::packetMonitorMain(uint32_t)':
WiFiScan.cpp:2007:31: error: 'class TFT_eSPI' has no member named 'getTouch'
     pressed = display_obj.tft.getTouch(&t_x, &t_y);

exit status 1
'class TFT_eSPI' has no member named 'setTouch'
`

tobozo, 2020-10-21:
`cp User_Setup.h /path/to/your/arduino/libraries/TFT_eSPI/User_Setup.h`

LuckyFishGeek, 2020-10-21:
@tobozo thank you for your help  

I will try it right away

closed 2020-10-21 by LuckyFishGeek