justcallmekoko/ESP32Marauder #59: How to make ttgo T8 V1.3 come with TF card interface to work？Ask for help

LuckyFishGeek, 2020-10-22:
TF card inserted into the interface on the screen can work,But in this case, the side of the screen will appear irregular and protrude.
[image]

When I plug in the ESP32 interface, the TF card does not work
Please help me, how can I set up the TF card of ESP32 to work

The following is the schematic diagram of my configuration settings and T8 V1.3

Please save the child
[image]
[image]

#define SD_CS 12
#define SD_MOSI 23
#define SD_MISO 2
#define SD_SCK 18

tobozo, 2020-10-22:
Can you try : 

  if (!SD.begin( 12 )) {
    Serial.println("Card Mount Failed");
  }

LuckyFishGeek, 2020-10-22:
> Can you try :
> 
> ```c
>   if (!SD.begin( 12 )) {
>     Serial.println("Card Mount Failed");
>   }
> ```

I'll try it right away, thanks for your help

LuckyFishGeek, 2020-10-22:
> Can you try :
> 
> ```c
>   if (!SD.begin( 12 )) {
>     Serial.println("Card Mount Failed");
>   }
> ```

The test failed. Is there any other way?

Thank you brother for your kind help

tobozo, 2020-10-22:
The help you get has the same quality as the feedback you give :-)

First make sure that the SD Card 

  - is *correctly* inserted (not fully inserted on the pic you posted)
  - is formatted FAT32
  - doesn't have a partition that exceeds 4GB
  - isn't faulty (e.g. can be accessed correctly from another device).

Then you can try to add a pinMode() before SD.begin

```C
    int SDCardCSPin = 12;

    pinMode( SDCardCSPin, OUTPUT );  // the SD library should take care of that.
    if (!SD.begin( SDCardCSPin )) {
      Serial.println("Card Mount Failed");
    }
```

If it still doesn't work, you may try other values for SDCardCSPin (e.g. 15 is the hardware CS pin for SPI)

Where did you get those values from ?

    #define SD_CS 12
    #define SD_MOSI 23
    #define SD_MISO 2
    #define SD_SCK 18

MOSI/MISO/SCK seem to have standard values, but any TTGO example I've found uses 15 for SD_CS, maybe this pinout is for another variant of the TTGO-T8 ?

Anyway, see how [this worked](https://forum.arduino.cc/index.php?topic=594910.0) with a different model of yours, the pin numbers may be different but the logical approach to find the pins is valid.

Happy hacking !

justcallmekoko, 2020-10-22:
It looks like the pin you have defined as `SD_CS` is not the proper pin that shows in the TTGO diagram you sent.

the CS pin of the SD card is 'DATA3' which looks like it is connected to `GPIO13` of the ESP32.

Also MOSI looks like it is connected to `GPIO15` and not `GPIO23` and SCK is connected to `GPIO14` and not `GPIO18`

So `#define SD_MISO 2` looks ok. You should just need to change `SD_MOSI`, `SD_SCK`, and `SD_CS`.

I am also curious where you got these definitions from.
```C++
#define SD_CS 12
#define SD_MOSI 23
#define SD_MISO 2
#define SD_SCK 18
```

LuckyFishGeek, 2020-10-22:
> It looks like the pin you have defined as `SD_CS` is not the proper pin that shows in the TTGO diagram you sent.
>
> the CS pin of the SD card is 'DATA3' which looks like it is connected to `GPIO13` of [...]

Thanks for your help, I will test as you said

LuckyFishGeek, 2020-10-22:
> The help you get has the same quality as the feedback you give :-)
>
> First make sure that the SD Card
>
> * is _correctly_ inserted (not fully inserted on the pic you posted)
> * is formatted FAT32
> * [...]

Thanks for your help, I will test as you said

LuckyFishGeek, 2020-10-22:
@tobozo @justcallmekoko 
Re-wired according to what you just said, but the TF card still does not work
Keep asking for help
Thank you two friends for helping me
Below is my wiring and configuration diagram

[image]
[image]
[image]
[image]

justcallmekoko, 2020-10-22:
Are you still inserting the SD card into the slot on the TTGO or are you putting in in the slot behind the screen?

LuckyFishGeek, 2020-10-22:
> Are you still inserting the SD card into the slot on the TTGO or are you putting in in the slot behind the screen?

Both methods have been tested and they don’t work

LuckyFishGeek, 2020-10-22:
> Are you still inserting the SD card into the slot on the TTGO or are you putting in in the slot behind the screen?

I will check the line again today to see if the connection is wrong

justcallmekoko, 2020-10-23:
If you use the SD slot behind the screen, I am confident the connection method [here](https://github.com/justcallmekoko/ESP32Marauder#connections) will work as long as you wire it accordingly. 

It looks like the pins in the I use for making the SD connection are still broken out on your TTGO board. This should work.

LuckyFishGeek, 2020-10-23:
> If you use the SD slot behind the screen, I am confident the connection method [here](https://github.com/justcallmekoko/ESP32Marauder#connections) will work as long as you wire it accordingly.
>
> It [...]

According to your wiring method, I can read the SD card from the interface on the screen the first time

Can the SD card be read on TTGO T8 V1.3 after optimization?

Thank you again for your design

justcallmekoko, 2020-10-23:
Yes but according to the diagram of the TTGO, the MISO, MOSI, SCK, and CS pins are different from what the marauder code expects. I am not sure how to to this, but you will need to specify those pins when calling `SD.begin()`

tobozo, 2020-10-23:
@LuckyFishGeek can you please try this sketch with a SD inserted in the TTGO onboard SD Card reader ?

```C

#include <SD.h>

#define SD_SCLK  14
#define SD_MISO   2
#define SD_MOSI  15
#define SD_CS    13

SPIClass *sdhander = nullptr;

bool sdcard_begin()
{
  if (!sdhander) {
    sdhander = new SPIClass(HSPI);
    sdhander->begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
  }
  if (!SD.begin(SD_CS, *sdhander, 20000000)) { // default = 20MHz but capped by SPI at 16MHz
    log_e("SD Card Mount Failed");
    return false;
  }
  return true;
}

void setup()
{
  Serial.begin(115200);
  if( ! sdcard_begin() ) {
    Serial.println("Can't read SD Card, aborting");
    while(1);
  } else {
    Serial.println("SD Card success !!");
  }
}

void loop() { }

```

You will have to momentarily remove the wires from the TFT until you find out the correct pinout for the TTGO onboard SD card reader.

/!\ Make sure you perform a full power cycle after every change in the `#define` values

Only when the valid pinout is found, re-wire the TFT in order to check if the SD card is still detected.

If your SD card starts failing after that, it probably means there are pin conflicts between the TFT and the SD.

If your SD Card is still working, you can use the pinout in the ESPMarauder and start testing on this basis.

Random tip: make sure you've set the debug level to "warn" before flashing.

[image]

LuckyFishGeek, 2020-10-23:
> @LuckyFishGeek can you please try this sketch with a SD inserted in the TTGO onboard SD Card reader ?
>
> ```c
> #include <SD.h>
>
> #define SD_SCLK  14
> #define SD_MISO   2
> #define SD_MOSI  15
> #define [...]

Okay, I will test it tomorrow. It is already 8 o'clock on my side and I am lying in bed.  Thank you again for your enthusiastic help.  I will feedback the test results in time

LuckyFishGeek, 2020-10-25:
> @LuckyFishGeek can you please try this sketch with a SD inserted in the TTGO onboard SD Card reader ?
>
> ```c
> #include <SD.h>
>
> #define SD_SCLK  14
> #define SD_MISO   2
> #define SD_MOSI  15
> #define [...]

[image]
[image]
After using your code, the SD card prompts success

The second picture is the pin diagram written on the board

tobozo, 2020-10-25:
yay !

This pinout is from the documentation you provided earlier, it's using hardware SPI.

[image]

Now see if this TFT pinout (taken from TFT_eSPI TTGO config) can coexist with your actual SD configuration, just wire the TFT to these pins, and check that the SD example sketch still detects your SD Card successfully : 

```
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    5
#define TFT_DC    0
#define TFT_RST   4
#define TOUCH_CS 13
```

If so, you can use these settings with the ESP32 Marauder and go on your WiFi duties :-)

LuckyFishGeek, 2020-10-26:
> yay !
>
> This pinout is from the documentation you provided earlier, it's using hardware SPI.
>
> [image]
>
> Now see if this TFT pinout (taken from TFT_eSPI TTGO config) can coexist with your actual SD [...]

Thank you for your kind help to me.  I will be going out a far door recently, and I will report the test results when I get home.  I wish you a happy life again

LuckyFishGeek, 2020-10-27:
@tobozo 
[image]
[image]
Seems to conflict

LuckyFishGeek, 2020-10-27:
@tobozo 
[image]

SD Card success !!

justcallmekoko, 2020-10-28:
This might be a @tobozo question, but I am pretty confident definitions in the main file supersede definitions in header files. You should be good.

Worst case scenario, I am wrong and you just have to try again with modified values in the header file as well.

LuckyFishGeek, 2020-11-07:
> This might be a @tobozo question, but I am pretty confident definitions in the main file supersede definitions in header files. You should be good.
> 
> Worst case scenario, I am wrong and you just have to try again with modified values in the header file as well.

Thanks for your kind help, I will test it tomorrow

LuckyFishGeek, 2020-11-08:
@tobozo @justcallmekoko  
Still unable to mount the SD card

This is my setting. Can you help me see if my setting is wrong? Now the screen can’t display normally!
thanks。。
[image]
[image]

user_Setup.h    I am using the default configuration, do I need to modify this file to be the same as SDinterface.h?
[image]

[image]

tobozo, 2020-11-08:
in your configuration SD_CS and TOUCH_CS are using the same pin, this is wrong.

[image]

Can you try to assign TOUCH_CS to a *free* pin ?

Also where is your User_Setup.h file ? is it still in ESPMarauder folder or did you overwrite the file in TFT_eSPI library folder as you're supposed to ?

LuckyFishGeek, 2020-11-09:
@tobozo I modified C:\Users\Administrator\Documents\Arduino\libraries\TFT_eSPI-master\User_Setup.h
[image]
Also modified
[image]
  not work
[image]
[image]
Thank you again for your help

tobozo, 2020-11-09:
TFT_RST (pin 12) is already used by the SD Card (data port), so you need to move this TFT_RST one to another free pin

[image]

LuckyFishGeek, 2020-11-09:
[image]

Synchronously modified C:\Users\Administrator\Documents\Arduino\libraries\TFT_eSPI-master\User_Setup.h
[image]

It can be displayed correctly, but the SD card still cannot be mounted!
This question is really stubborn. Haha
[image]
About SD card
I tried the 16G and 1G capacity of SanDisk brand
Format to FAT32

All show failed to mount
@tobozo

tobozo, 2020-11-09:
test : 

- unplug the TFT Wires one after the other until the SD Card works
- when the SD works : you found the conflicting pin !

LuckyFishGeek, 2020-11-10:
> test :
> 
> * unplug the TFT Wires one after the other until the SD Card works
> * when the SD works : you found the conflicting pin !

@tobozo 
It's weird. I tested it as you said, every time I unplug a pin and a wire, I restart it! Then see if the SD card is mounted!
As a result, all the pins have been tested, and the SD card prompts that it cannot be mounted!

tobozo, 2020-11-10:
> I unplug a pin and a wire (...) Then see if the SD card is mounted!

>  all the pins have been tested, and the SD card prompts that it cannot be mounted!

Those are conflicting informations.

- when does it mount, and when does it not mount ?
- how many TFT pins are unplugged when the SD card mounts successfully, and which are they ?

Please be methodic and specific otherwise I can't provide you with efficient help.

LuckyFishGeek, 2020-11-10:
@tobozo 
Maybe I have a problem with the previous expression, sorry
My operation is as follows: Unplug the pins on the TFT screen one by one, and every time I unplug a pin, I will restart once to see if the SD card is successfully mounted
CS
RESET
D/C
MOSI
SCK
led
MISO
T_CLK
T_CS
T_DI
T_DO

[image]
[image]
Just like this, I now unplug the TFT_CS pin, then I click reset and check if the SD card is mounted!
Is my test method correct? Or do you need to unplug all the TFT screen pins at once?

tobozo, 2020-11-10:
Thanks for this feedback, it sounds like you're very close :-)

Some thoughts:

The message `IP5306 I2C supported: false` means some Wire.begin() occured, which you probably don't want. 
From the BatteryInterface.h file: 

    #define I2C_SDA 33
    #define I2C_SCL 22

You can eventually comment out this line in the .ino file to make sure this is not interferring with the TFT:

    // battery_obj.RunSetup();

LuckyFishGeek, 2020-11-11:
@tobozo 
Good morning I'm only 10 am here, I don't know what time it is over there, haha

According to what you said, I commented out this, after re-flashing the firmware, the same error still appeared
[image]

tobozo, 2020-11-11:
Well that's the difference between "helping" and "assisting", providing some methodologies, not solutions.

I'm sure you'll eventually find the correct pins combination by yourself, happy hacking !

closed 2020-11-11 by LuckyFishGeek