justcallmekoko/ESP32Marauder #60: not compile on arduino ide

rferrari81, 2020-10-28:
Arduino:1.8.13 (Windows 10), Scheda:"ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Più di una libreria trovata per "SD.h"

In file included from c:\users\rferr\documents\arduino\libraries\lvgl\src/lv_misc/lv_log.h:16:0,

Usata: C:\Users\rferr\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD

                 from c:\users\rferr\documents\arduino\libraries\lvgl\lvgl.h:26,

Non usata: C:\Program Files (x86)\Arduino\libraries\SD

                 from C:\Users\rferr\Documents\Arduino\libraries\lvgl\src/lvgl.h:17,

Non usata: C:\Users\rferr\Documents\Arduino\libraries\SD

                 from sketch\Display.h:11,

Più di una libreria trovata per "WiFi.h"

                 from C:\Users\rferr\Desktop\ESP32Marauder-master\esp32_marauder\esp32_marauder.ino:20:

Usata: C:\Users\rferr\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi

c:\users\rferr\documents\arduino\libraries\lvgl\src\lv_conf_internal.h:24:101: fatal error: ../../lv_conf.h: No such file or directory

Non usata: C:\Program Files (x86)\Arduino\libraries\WiFi

compilation terminated.

Più di una libreria trovata per "SPI.h"

Usata: C:\Users\rferr\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI

Non usata: C:\Users\rferr\Documents\Arduino\libraries\SPI

Più di una libreria trovata per "Wire.h"

Usata: C:\Users\rferr\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire

Non usata: C:\Users\rferr\Documents\Arduino\libraries\Wire

exit status 1

Errore durante la compilazione per la scheda ESP32 Dev Module.

Gotcha007, 2021-02-21:
I have kind of the same. To be honest, it's the first time playing with Arduino.

Multiple libraries were found for "SD.h"
In file included from c:\users\gotch\documents\arduino\libraries\lvgl\src/lv_misc/lv_log.h:16:0,
 Used: C:\Users\gotch\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\SD
                 from c:\users\gotch\documents\arduino\libraries\lvgl\lvgl.h:25,
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\SD
Multiple libraries were found for "lvgl.h"
 Used: C:\Users\gotch\Documents\Arduino\libraries\lvgl
 Not used: C:\Users\gotch\Documents\Arduino\libraries\lv_arduino
Multiple libraries were found for "WiFi.h"
                 from C:\Users\gotch\Documents\Arduino\libraries\lvgl\src/lvgl.h:17,
 Used: C:\Users\gotch\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
                 from sketch\Display.h:11,
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi
                 from C:\Users\gotch\Downloads\ESP32Marauder-master\ESP32Marauder-master\esp32_marauder\esp32_marauder.ino:20:
c:\users\gotch\documents\arduino\libraries\lvgl\src\lv_conf_internal.h:40:101: fatal error: ../../lv_conf.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board LOLIN D32.

astronomicalunit, 2021-02-21:
man that is brutal it sounds like a Jenkins_Exploit to take any data or hpothetucal proprietary info to if u email or talk of projects with dSTIKE folks some in china, i heard the gov there is searchings fior anyone cresting things that meet their extremely broad law list of ways to be a dissident like jaywalking. but also could be AWS cloud trail monitoring by consensu.org aws Lambada, Rda real data base intaces, aws+googles cloud watch groups for attrbutes names code in AwS SNS as they watch github. or a mirkataz attack looks like too but the code cmds look like AWS IAmAccessKeys IAMSecretAcessKeys with S3,
the IambGroups,policy, managed policies and the outh is in each command even book 'operator hand book search cooypast if known blue red hardening, hacking, and defense attack or tip is listed for all devices os etc... but that one the first code uses github and reddit or "code site" like (iam list-all 'domain-query target) -all-acesskeys) or iam, user i am group iam secret user iam receiving all interactions, or aws-ec2 describe target secret-keys describe-images-geotag-all 'domain' query or region-describe all behavioral traits, decribe-regions-encypted-message-photo-copy/paste-enail, ip-hobbies-describe-hundreds more and at end ad reddit, github tindie, and region anyone say sharing code with aws ec2 describe-region China-russian-iran- etc -describe ex. query code tech, word spoof, for describe-5eyes-citizen describe secrete keys OAuth accounts witt fam or friend with hobby similar describe regions china us dats packets GitHub etc.. trippy. ill look into it more. excuse errors using talk to text. hope all is well this weekend

. its def been a rough year. some tried asking me if i can find a us citizen or possible ex intell agent woking for fbi or cia, one i sold the CAINE fed software too. he was chill but said he might be connected or an informant, so i said i could but ill think on it. as i got a weird feeling. day after all my githubs IG got authed accessed somehow. 33 character pw but ive only used it to get stoked on your projects i dont even have a repos or anything. just one comment saying your project is great and mentioned with consent like in red/blue team oe to test in your own network, its such an incredible tool.. but back to the other guy, i sent him a message saying although u have beeb nice and showed me a PI license, him being a possible cia fbi mid level assets makes it illegal in my state as i dont even have a license.. which he said "great job u passed, im actually fbi special agent... and i finished by saying his name. he offered a job snd said by the way you selling those toola via usb, arent illegal but say i lived in example state, and somehow u bought these from a former fed as the one i got i didnot belive had admin fed credtial pw to literally tap phones entire 3 federal print crimibal and behaviavioral data bases.. he was like im not giving u trouble but make any others without and have sign a waiver as just selling to me was a strike maybe 2-5 years fed time.. if they user for crime finacial gain, ud have got 20-30.. bur ur a good kid and know u have about 850 of 1,006 followers from every state and fed agency and u called them out so we thought u had a clone of pegasus. i do but buried in a diff state.. its so powerful im afraid to use it. it cost 110 million and 55k month to govs who buy it. the verision i got apple sec citizen lab and others hadnt seen a reg us citizen with it on phone.. they were so scared as a day earlier they mentioned patching okd version yet i found showed they as in NSO and ex TAO who contract in saudi making tools for a regime with mass human rights targeted me, plus i told NSO i have their new code, which is on ur devices and full ACXIOM (huge data company that 10 yrs ago had 3,000 or more attributes each pages long detail on every citizen in us uk ca mex but today if one uses any android, blu tooth in car, or walks 6 ft into any shop isle where beacons get a frequency YALE MIT plus eff.org and dozens of adocates found is on every phone.,24/7, even in airplane mode, turned off, settibgs have it off, and it range of 65 ft we dont hear it, can do a lot. it connects secretky to any device in home, even if left 3rd floor u in basemenf on work air gapped device it trails from tv to watch to u or a window, which colkects vibrations from typing, even pen writing, and can get contacts message, to device even using hotspots or conmunicating to airgap machine to silently turn odf features every tv show to porn even health as it can literallly hear your organs which is sold to say axciom finspy and scarier Futures which added ti another word like axiom today has 50,000 attributes on 3 billion people. 100,000 on every us legal non legal and Ally and by dybyour born has the second u took first breath to a prediction algorriom used by darpa in their A.W.A.R.E and TIA programa whicu disturbed even cia directorbat time as 95% accuracy predicting how when from what soneone dies, to using paerens from grade school to predict score and label sell to DOJ ex, as to be used in sentencing as say mom single broke, u eat cheap food, or drink out of plastic, to porn habit, tia stabding for total information program which transfered to Saad/AxCiom with furtures, and u can order ur data but unless u habe finspy or peg which version i have takes 37-44 days to infect every device in a smaller country and 25 to do west coast 56 for usa. finspy factors from a point [... cut ...] ch like that as it unkike most can be offensive, but enhanced with teams labs to be a accesory sold only in certain nations lijke cars here. ontaining intel, the night stand2 could be both bsttery and 1000 yard packet injection to modified wifi pineapple by creating a hidden mesh network, device to device, to smalll jammers that kids in middle east in school wear that told its to stay off ohone etc, but prevents singals for remote ieds as schools are buggesr tsrget..

Sent from ProtonMail Mobile

justcallmekoko, 2021-02-22:
@Gotcha007 

> from C:\Users\gotch\Downloads\ESP32Marauder-master\ESP32Marauder-master\esp32_marauder\esp32_marauder.ino:20:
> c:\users\gotch\documents\arduino\libraries\lvgl\src\lv_conf_internal.h:40:101: fatal [...]

It sounds like you need to install the lv_arduino library, not lvgl. If you go to the library manager in the Arduino IDE, you will find lv_arduino. You can install it there. One of the differences I noticed between the two libraries is one included an lv_conf.h and the other does not.

Gotcha007, 2021-02-22:
Thanks for that, I had it installed but I think the issue might have been that I had both installed. I removed Lvgl and after a few errors I was able to resolve, I'm getting this now.

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

WiFiScan.cpp:1141:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

         DynamicJsonBuffer jsonBuffer;

           ^

sketch\Display.cpp: In member function 'void Display::RunSetup()':

Display.cpp:42:7: error: 'class TFT_eSPI' has no member named 'setTouch'

   tft.setTouch(calData);

       ^

sketch\Display.cpp: In member function 'void Display::drawStylus()':

Display.cpp:482:25: error: 'class TFT_eSPI' has no member named 'getTouch'

   boolean pressed = tft.getTouch(&x, &y);

                         ^

sketch\MenuFunctions.cpp: In function 'bool my_touchpad_read(lv_indev_drv_t*, lv_indev_data_t*)':

MenuFunctions.cpp:47:34: error: 'class TFT_eSPI' has no member named 'getTouch'

   bool touched = display_obj.tft.getTouch(&touchX, &touchY, 600);

                                  ^

sketch\MenuFunctions.cpp: In member function 'void MenuFunctions::main(uint32_t)':

MenuFunctions.cpp:322:29: error: 'class TFT_eSPI' has no member named 'getTouch'

   pressed = display_obj.tft.getTouch(&t_x, &t_y);

                             ^

sketch\MenuFunctions.cpp: In member function 'void MenuFunctions::orientDisplay()':

MenuFunctions.cpp:668:19: error: 'class TFT_eSPI' has no member named 'setTouch'

   display_obj.tft.setTouch(calData);

                   ^

sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunLvJoinWiFi(uint8_t, uint16_t)':

WiFiScan.cpp:439:19: error: 'class TFT_eSPI' has no member named 'setTouch'

   display_obj.tft.setTouch(calData);

                   ^

sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunPacketMonitor(uint8_t, uint16_t)':

WiFiScan.cpp:604:19: error: 'class TFT_eSPI' has no member named 'setTouch'

   display_obj.tft.setTouch(calData);

                   ^

sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunEapolScan(uint8_t, uint16_t)':

WiFiScan.cpp:651:19: error: 'class TFT_eSPI' has no member named 'setTouch'

   display_obj.tft.setTouch(calData);

                   ^

sketch\WiFiScan.cpp: In static member function 'static void WiFiScan::pwnSnifferCallback(void*, wifi_promiscuous_pkt_type_t)':

WiFiScan.cpp:1141:27: error: 'jsonBuffer' was not declared in this scope

         DynamicJsonBuffer jsonBuffer;

                           ^

WiFiScan.cpp:1143:19: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6173_91::ObjectRef}' has no member named 'success'

         if (!json.success()) {

                   ^

WiFiScan.cpp:1150:16: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6173_91::ObjectRef}' has no member named 'printTo'

           json.printTo(json_output);

                ^

sketch\WiFiScan.cpp: In member function 'void WiFiScan::eapolMonitorMain(uint32_t)':

WiFiScan.cpp:1774:31: error: 'class TFT_eSPI' has no member named 'getTouch'

     pressed = display_obj.tft.getTouch(&t_x, &t_y);

                               ^

sketch\WiFiScan.cpp: In member function 'void WiFiScan::packetMonitorMain(uint32_t)':

WiFiScan.cpp:2007:31: error: 'class TFT_eSPI' has no member named 'getTouch'

     pressed = display_obj.tft.getTouch(&t_x, &t_y);

                               ^

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\gotch\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi

Multiple libraries were found for "SD.h"

 Used: C:\Users\gotch\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\SD

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\SD

exit status 1

DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

justcallmekoko, 2021-02-22:
@Gotcha007 

> DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

Go back to library manager and install the ArduinoJson version RIGHT BEFORE version 6. It should be version 5.x.x
You just have to make sure it isn't 6.x.x

As for the other errors, this is an issue I need to resolve with the marauder code itself as the current version of TFT_eSPI no longer uses setTouch and getTouch. I am still using an old version of the library. I have a branch of TFT_eSPI that should still work if you would like to try installing that instead

tobozo, 2021-02-22:
I apologize in advance for the dummy question, but why is the old ArduinoJSON needed in the first place ?

What about PR #56, the code change does not look that big, is that a workable solution ?

Gotcha007, 2021-02-22:
> @Gotcha007
>
>
> Go back to library manager and install the ArduinoJson version RIGHT BEFORE version 6. It should be version 5.x.x
> You just have to make sure it isn't 6.x.x
>
> As for the other errors, [...]

That resolved that issue, thanks !
Can you share your library for the TFT then. I used that one but still getting the errors
https://github.com/justcallmekoko/TFT_eSPI

justcallmekoko, 2021-02-22:
Which errors are you receiving when you use my branch?

@tobozo The reason I used the ArduinoJson solution I use know is because it was the most consistent answer I found

Gotcha007, 2021-02-22:
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

sketch\Display.cpp: In member function 'void Display::RunSetup()':

Display.cpp:42:7: error: 'class TFT_eSPI' has no member named 'setTouch'

   tft.setTouch(calData);

       ^

sketch\Display.cpp: In member function 'void Display::drawStylus()':

Display.cpp:482:25: error: 'class TFT_eSPI' has no member named 'getTouch'

   boolean pressed = tft.getTouch(&x, &y);

                         ^

sketch\MenuFunctions.cpp: In function 'bool my_touchpad_read(lv_indev_drv_t*, lv_indev_data_t*)':

MenuFunctions.cpp:47:34: error: 'class TFT_eSPI' has no member named 'getTouch'

   bool touched = display_obj.tft.getTouch(&touchX, &touchY, 600);

                                  ^

sketch\MenuFunctions.cpp: In member function 'void MenuFunctions::main(uint32_t)':

MenuFunctions.cpp:322:29: error: 'class TFT_eSPI' has no member named 'getTouch'

   pressed = display_obj.tft.getTouch(&t_x, &t_y);

                             ^

sketch\MenuFunctions.cpp: In member function 'void MenuFunctions::orientDisplay()':

MenuFunctions.cpp:668:19: error: 'class TFT_eSPI' has no member named 'setTouch'

   display_obj.tft.setTouch(calData);

                   ^

sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunLvJoinWiFi(uint8_t, uint16_t)':

WiFiScan.cpp:439:19: error: 'class TFT_eSPI' has no member named 'setTouch'

   display_obj.tft.setTouch(calData);

                   ^

sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunPacketMonitor(uint8_t, uint16_t)':

WiFiScan.cpp:604:19: error: 'class TFT_eSPI' has no member named 'setTouch'

   display_obj.tft.setTouch(calData);

                   ^

sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunEapolScan(uint8_t, uint16_t)':

WiFiScan.cpp:651:19: error: 'class TFT_eSPI' has no member named 'setTouch'

   display_obj.tft.setTouch(calData);

                   ^

sketch\WiFiScan.cpp: In member function 'void WiFiScan::eapolMonitorMain(uint32_t)':

WiFiScan.cpp:1774:31: error: 'class TFT_eSPI' has no member named 'getTouch'

     pressed = display_obj.tft.getTouch(&t_x, &t_y);

                               ^

sketch\WiFiScan.cpp: In member function 'void WiFiScan::packetMonitorMain(uint32_t)':

WiFiScan.cpp:2007:31: error: 'class TFT_eSPI' has no member named 'getTouch'

     pressed = display_obj.tft.getTouch(&t_x, &t_y);

                               ^

Multiple libraries were found for "SD.h"

 Used: C:\Users\gotch\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\SD

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\SD

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\gotch\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi

exit status 1

'class TFT_eSPI' has no member named 'setTouch'

Gotcha007, 2021-02-22:
Which is odd because I do find the function in Touch.h file

justcallmekoko, 2021-02-22:
It makes me think you might still have the original TFT_eSPI library installed. If you do to the libraries directory in your Arduino folder, you should be able to delete the TFT_eSPI library then install my branch. I don't know if this is what you already did, but if it is, there is an actual ghost in your computer

Gotcha007, 2021-02-22:
So I removed the library and compile. got that error which is normal.
TFT_eSPI.h: No such file or directory
so that means that nothing else exists.

I put back the one from your repository and still getting that error. Very stange. Do I have to do something when I install a library?

Gotcha007, 2021-02-22:
uninstalled, deleted the cache and the arduino folder into the document folder. re-install everything and made sure to only install your library for TFT_eSPI but still having the same issue :-(

Gotcha007, 2021-02-22:
I'm wondering if it's because it never loads this (from TFT_eSPI.h)
// Load the Touch extension
#ifdef TOUCH_CS
  #include "Extensions/Touch.h"        // Loaded if TOUCH_CS is defined by user
#endif

Gotcha007, 2021-02-22:
I feel stupid but do I need to have the screen already connected to have it compiled correctly? I only have the board connected at the moment and did not do the wiring

Gotcha007, 2021-02-23:
ok I was finally able to fix my issues by following this
https://github.com/Bodmer/TFT_eSPI/issues/454#issuecomment-548085768

and also this
https://github.com/justcallmekoko/ESP32Marauder/issues/2#issuecomment-783735316

Now I'm getting the below. Am I doing something wrong?

Arduino: 1.8.13 (Windows 10), Board: "LOLIN D32 PRO, Disabled, Default, 80MHz, 921600, None"

text section exceeds available space in board

Sketch uses 1460257 bytes (111%) of program storage space. Maximum is 1310720 bytes.

Global variables use 107252 bytes (32%) of dynamic memory, leaving 220428 bytes for local variables. Maximum is 327680 bytes.

Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.

Error compiling for board LOLIN D32 PRO.

justcallmekoko, 2021-02-23:
Set your partition scheme to Large app and Small SPIFS or something like that.

Gotcha007, 2021-02-23:
You are the best, thanks it worked !
now I need to wire that screen lol
I'm still planning to buy one of yours anyway (in the wait list) but wanted to play with it

justcallmekoko, 2021-02-25:
@rferrari81 
> c:\users\rferr\documents\arduino\libraries\lvgl\src\lv_conf_internal.h:24:101: fatal error: ../../lv_conf.h: No such file or directory

You are having the same issue. You need to uninstall the lvgl library in the Arduino IDE and install lv_arduino instead.

@Gotcha007 let me know how it goes with the screen wiring. I would love to see the finished product

closed 2021-02-25 by justcallmekoko

Gotcha007, 2021-02-25:
Yep, I will !
To be honest I'm struggling a little to know how to wire the Lolin D32 Pro to the TFT screen.
First time doing such thing, I usually only play with Raspberries.