justcallmekoko/ESP32Marauder #61: analog battery circuit

rferrari81, 2020-10-28:
hi everyone, i would like to know how the analog battery circuit is connected and how it is done thanks.

justcallmekoko, 2020-10-28:
@Joshinken

Joshinken, 2020-10-29:
WHAT, I WASNT SLEEPING IN CLASS, IM AWAKE

@rferrari81 it uses a voltage divider, with the wiring being battery->resistor->gpio34->resistor->ground. If you want to reduce power usage you can either connect an n-channel mosfet between the second resistor and ground or a p-channel mosfet between the battery and the first resistor and connect the gate of the mosfet to gpio13 (otherwise the circuit will draw very tiny amounts of power continously, even in standby, which might damage your battery if you dont use an undervoltage protection circuit). The values of the resistors themselves dont matter as long as the first resistor is 4 times the size of the second, however i went with 4KΩ and 1KΩ.

rferrari81, 2020-10-29:
You dont't have circuit Image and MOSFET type? Thanks

Joshinken, 2020-10-29:
I dont have a schematic because its a very simple circuit and can easily be changed in a lot of different ways and still work the exact same. If u were to make a schematic it would just be a picture of 2 resistors labeled “n” and “4n”

I told you the type of mosfet and where to put which type of mosfet. If youre asking for a model number for the mosfet, dont, because any mosfet you have lying around will do the trick

closed 2020-10-30 by justcallmekoko