justcallmekoko/ESP32Marauder #65: scan for hidden cameras

paulsijben, 2020-11-11:
seeing how easy unscrupulous airbnb owners can buy hidden cameras, it would be so useful to scan for these devices by mac or ssid? (e.g. https://www.fing.com/news/how-to-find-hidden-cameras-in-your-airbnb)

justcallmekoko, 2020-12-03:
@paulsijben Sorry I haven't gotten back to you regarding this issue. A month is a long time to wait for a response so I appreciate you being patient. 

I think this would be very possible to implement. It would be as simple as creating a function similar to the "scan for espressif" tool but rather than using a list of known espressif mac addrs, it would use a list of known camera mac addrs.

As for scanning the area for networks with a suspicious name such as "Hidden Cam", this can be accomplished with the generic SSID Scan tool that is already on the marauder.

If you are so inclined, you can develop the first method I discussed on your own by copying the functionality of the espressif scan and I will include it in the next firmware update.

If you are not inclined, I can work on getting this done myself. It just might take longer because I have a lot on my plate at the moment.

paulsijben, 2020-12-03:
No problem, I was not sleeping anywhere else than at home these days... 
I'd be happy to take a stab at writing something like that. 

Is someone reading this who has a list of mac prefixes for common hidden/spy cams?

paulsijben, 2020-12-03:
sorry, wrong button

justcallmekoko, 2021-06-07:
Found some lists of manufacturer camera OUIs [here](https://ipvm.com/forums/video-surveillance/topics/is-there-a-list-of-the-top-camera-manufacturer-s-mac-ouis)

Need to do some more research.

justcallmekoko, 2021-06-07:
The ESP32 is not capable of storing a manufacturer list large enough for a hidden camera scanner to be effective. There is not enough memory available.

closed 2021-06-07 by justcallmekoko