justcallmekoko/ESP32Marauder #68: digikey pyportal

one0eight, 2020-12-30:
Is there any reason this cant be flashed to a digikey pyportal?

tobozo, 2020-12-30:
do you mean it can't or are you asking ?

justcallmekoko, 2020-12-30:
@tobozo +1
@one0eight are you claiming it cannot and you want to know why it is not possible or are you asking why no one has tried?

I have not tried this but judging by the description, it sounds like the display is driven by the ATSAMD51J20. If this is the case, the current ESP32 Marauder code would not function properly. Since the esp32 is the co-processor, the display potions of this code and the wifi portions would need to be separated into two parts and flashed accordingly.

But the idea of some pentesting tools running on this is actually pretty cool considering the ATSAMD51J20 can function as a HID. This would mean that you can run keystroke injection attacks.

closed 2020-12-30 by justcallmekoko