justcallmekoko/ESP32Marauder #69: Rotate all display to work in landscape for Marauder Centauri

justcallmekoko, 2021-01-13:
[Marauder Centauri's](https://github.com/justcallmekoko/ESP32Marauder/tree/MarauderCentauri) screen is rotated in landscape mode. I can't think of the direction off the top of my head but the ribbon cable is on the right side of the screen now.

The height, width, and likely the button sizes will need to be changed.

tobozo, 2021-01-13:
also this impacts hardware scrolling depending on the display model+version

For example M5Stack ILI9341 is set for vertical *landscape* scrolling while some ST7789 can only do vertical *portrait* scrolling.

justcallmekoko, 2021-01-13:
@tobozo I heavily suspected that and I am not looking forward to trying to fix that especially because I don't fully understand how it works in the first place.

If this cannot do landscape scrolling then this will complicate things considerably.

tobozo, 2021-01-13:
there's a way to use 1bit sprite (2 colors) and do software scrolling instead, it's not as smooth as hardware scrolling but still very usable for text only content.

justcallmekoko, 2021-01-13:
gotcha. How does lvgl accomplish their scrolling? With the current marauder functionality, it have a text box in landscape orientation and it scroll properly.

tobozo, 2021-01-13:
given the nature of the library I'd say it's software scroll

justcallmekoko, 2021-03-25:
Just realized this is on the wrong repo

closed 2021-03-25 by justcallmekoko