justcallmekoko/ESP32Marauder #71: 'class TFT_eSPI' has no member named 'setTouch'

RChadwick7, 2021-02-28:
I have this same issue. I configured Bodmer's library by commenting and uncommenting the proper lines in User_Setup_Select.h, and I uncommented line 50 in Display.h. Is there anything else I'm missing?

_Originally posted by @RChadwick7 in https://github.com/justcallmekoko/ESP32Marauder/issues/2#issuecomment-787115797_

justcallmekoko, 2021-03-01:
Did you get TFT_eSPI directly from bodmer's repo or did you install my branch of the repo?

Quas7, 2021-03-02:
Hi, I used the linked repo for the TFT_eSPI as described in the Arduino-IDE installation instruction

```Bibliothek TFT_eSPI-master in Version 2.0.2 im Ordner: C:\Users\tillm\Documents\Arduino\libraries\TFT_eSPI-master wird verwendet```

But I also run into this issue

```exit status 1
'class TFT_eSPI' has no member named 'setTouch'```

closed 2021-03-02 by justcallmekoko