justcallmekoko/ESP32Marauder #73: Compatibility with Esp32 TTGo T-Display

huukhoa1412, 2021-03-13:
Compatibility with Esp32 TTGo T-Display ?

justcallmekoko, 2021-03-24:
It should be compatible. You should only need to change the TFT_eSPI configuration to reflect the correct pins use by the TTGo T-Display

huukhoa1412, 2021-03-28:
Compile : 

sketch\Display.cpp: In member function 'void Display::RunSetup()':
Display.cpp:42:7: error: 'class TFT_eSPI' has no member named 'setTouch'
   tft.setTouch(calData);
       ^
sketch\Display.cpp: In member function 'void Display::drawStylus()':
Display.cpp:482:25: error: 'class TFT_eSPI' has no member named 'getTouch'
   boolean pressed = tft.getTouch(&x, &y);
                         ^
sketch\Web.cpp: In member function 'void Web::setupOTAupdate()':
Web.cpp:72:53: error: cannot convert 'esp_interface_t' to 'wifi_interface_t' for argument '1' to 'esp_err_t esp_wifi_set_mac(wifi_interface_t, const uint8_t*)'
   esp_wifi_set_mac(ESP_IF_WIFI_AP, &newMACAddress[0]);
                                                     ^
sketch\MenuFunctions.cpp: In function 'bool my_touchpad_read(lv_indev_drv_t*, lv_indev_data_t*)':
MenuFunctions.cpp:47:34: error: 'class TFT_eSPI' has no member named 'getTouch'
   bool touched = display_obj.tft.getTouch(&touchX, &touchY, 600);
                                  ^
sketch\MenuFunctions.cpp: In member function 'void MenuFunctions::main(uint32_t)':
MenuFunctions.cpp:322:29: error: 'class TFT_eSPI' has no member named 'getTouch'
   pressed = display_obj.tft.getTouch(&t_x, &t_y);
                             ^
sketch\MenuFunctions.cpp: In member function 'void MenuFunctions::orientDisplay()':
MenuFunctions.cpp:668:19: error: 'class TFT_eSPI' has no member named 'setTouch'
   display_obj.tft.setTouch(calData);
                   ^
WiFiScan.cpp:1141:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6
         DynamicJsonBuffer jsonBuffer;
           ^
sketch\WiFiScan.cpp: In member function 'String WiFiScan::getStaMAC()':
WiFiScan.cpp:380:63: error: cannot convert 'esp_interface_t' to 'wifi_interface_t' for argument '1' to 'esp_err_t esp_wifi_get_mac(wifi_interface_t, uint8_t*)'
   esp_err_t mac_status = esp_wifi_get_mac(ESP_IF_WIFI_STA, mac);
                                                               ^
sketch\WiFiScan.cpp: In member function 'String WiFiScan::getApMAC()':
WiFiScan.cpp:404:62: error: cannot convert 'esp_interface_t' to 'wifi_interface_t' for argument '1' to 'esp_err_t esp_wifi_get_mac(wifi_interface_t, uint8_t*)'
   esp_err_t mac_status = esp_wifi_get_mac(ESP_IF_WIFI_AP, mac);
                                                              ^
sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunLvJoinWiFi(uint8_t, uint16_t)':
WiFiScan.cpp:439:19: error: 'class TFT_eSPI' has no member named 'setTouch'
   display_obj.tft.setTouch(calData);
                   ^
sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunPacketMonitor(uint8_t, uint16_t)':
WiFiScan.cpp:604:19: error: 'class TFT_eSPI' has no member named 'setTouch'
   display_obj.tft.setTouch(calData);
                   ^
sketch\WiFiScan.cpp: In member function 'void WiFiScan::RunEapolScan(uint8_t, uint16_t)':
WiFiScan.cpp:651:19: error: 'class TFT_eSPI' has no member named 'setTouch'
   display_obj.tft.setTouch(calData);
                   ^
sketch\WiFiScan.cpp: In static member function 'static void WiFiScan::pwnSnifferCallback(void*, wifi_promiscuous_pkt_type_t)':
WiFiScan.cpp:1141:27: error: 'jsonBuffer' was not declared in this scope
         DynamicJsonBuffer jsonBuffer;
                           ^
WiFiScan.cpp:1143:19: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6173_91::ObjectRef}' has no member named 'success'
         if (!json.success()) {
                   ^
WiFiScan.cpp:1150:16: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6173_91::ObjectRef}' has no member named 'printTo'
           json.printTo(json_output);
                ^
sketch\WiFiScan.cpp: In member function 'void WiFiScan::eapolMonitorMain(uint32_t)':
WiFiScan.cpp:1774:31: error: 'class TFT_eSPI' has no member named 'getTouch'
     pressed = display_obj.tft.getTouch(&t_x, &t_y);
                               ^
sketch\WiFiScan.cpp: In member function 'void WiFiScan::packetMonitorMain(uint32_t)':
WiFiScan.cpp:2007:31: error: 'class TFT_eSPI' has no member named 'getTouch'
     pressed = display_obj.tft.getTouch(&t_x, &t_y);
                               ^
Multiple libraries were found for "SD.h"
 Used: C:\Users\LEHUUKHOA\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\LEHUUKHOA\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'class TFT_eSPI' has no member named 'setTouch'

justcallmekoko, 2021-06-07:
@huukhoa1412 make sure you are using my branch of the TFT_eSPI library

justcallmekoko, 2021-06-16:
@huukhoa1412 If you are using the proper branch, make sure you have properly configured your `User_Setup.h` file to the display you are using.