justcallmekoko/ESP32Marauder #74: Compatibility with TTGO T-Watch 2020

letoubib, 2021-03-24:
hello there
great work here
i'm very new to this , 
i tried ur code on TTGO T-Watch 2020 , had to make many changes to the TFT_eSPI setup , as well as to web.cpp, wifiscan.cpp , among others,  but still no go .. i get the error : 
`sketch\MenuFunctions.cpp: In member function 'void MenuFunctions::addNodes(Menu*, String, uint16_t, Menu*, int, std::function<void()>)':
MenuFunctions.cpp:1062:69: error: too many initializers for 'MenuNode'
   menu->list->add(MenuNode{name, color, place, &new_button, callable});
                                                                     ^
MenuFunctions.cpp:1062:69: error: could not convert '& new_button' from 'TFT_eSPI_Button*' to 'std::function<void()>'
In file included from C:\Users\xxx\Desktop\ESP32Marauder-master\esp32_marauder\esp32_marauder.ino:25:0:
Display.h:110:5: error: 'TFT_eSPI_Button' does not name a type
     TFT_eSPI_Button key[BUTTON_ARRAY_LEN];
`

justcallmekoko, 2021-03-24:
It sounds like you do not have the TFT_eSPI_Button library installed. This is issue on my part as I did not include this in the documentation

letoubib, 2021-03-24:
thanks for the reply 
forgive my noobs qs
i tried to look for this TFT_eSPI_Button library but couldnt find it , is there a link ???
thanks again

justcallmekoko, 2021-04-14:
This is my mistake. TFT_eSPI_Button is part of the TFT_eSPI library. I forget this very often because I only use the Button for one thing in marauder. Did you install the TFT_eSPI library from the github repo or did you install it from my branch in the README?

justcallmekoko, 2021-06-07:
Just realized this is a duplicate of #25

closed 2021-06-07 by justcallmekoko