justcallmekoko/ESP32Marauder #78: Display.h:7:25: fatal error: JPEGDecoder.h: No such file or directory

xcs011, 2021-04-27:
After doing a little searching, the mentioned file is not in the source code, or either prerequisite library. 
I appreciate any help you can give, thank you.

xcs011, 2021-04-27:
Ive also realized I have a hiletgo esp32-wroom, not a lolin, and cant find it in the board manager, although that isnt a problem with verification.

justcallmekoko, 2021-04-28:
This is my mistake. I did not document as I went along and there are a lot of steps I had forgotten about which are required to setup the dev environment.

You have to install [this library](https://github.com/Bodmer/JPEGDecoder). This should solve your issue. The ESP32 board should not matter. The main concern is that the board you select in the IDE must include the partition scheme to allow for large app and small spiffs.

Let me know if this resolves you issue.

xcs011, 2021-04-28:
That worked, thank you, the only two other errors I had were for linkedlist, and lvgl, but they were easily installed from the library manager.

xcs011, 2021-04-28:
lvhl appears to be fixed after configuring it, but I might have installed a bare bones arduino client or something, because im also misssing arduinojson.h

xcs011, 2021-04-28:
list of missing libraries for all interested:
lvgl
linkedlist
adafruit neopixel
arduinojson 5
nimblearduino

justcallmekoko, 2021-04-28:
Don't install lvgl. Install lv_arduino instead.

xcs011, 2021-04-28:
Thank you, the only two other errors I had were no set_touch, and out of space, but viewing closed issues helped, so now it compiles.
I will breadboard it and test it out.

justcallmekoko, 2021-04-28:
Awesome. Glad to hear you got it working. Post pictures when you get it put together. I would love to see it

closed 2021-04-28 by justcallmekoko