justcallmekoko/ESP32Marauder #79: A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x00) SPIFFS Upload failed!

xcs011, 2021-04-28:
SPIFFS keeps giving me this error upon upload.
My setup uses an external SD shield, connected to power and the necessary pinout pins, would this be solved by using the sd reader inside the display instead?

xcs011, 2021-04-28:
Asking instead of testing because I want to be sure before I go soldering anything

justcallmekoko, 2021-04-28:
I would use the SD reader in the display to keep it as consistent with the original design as possible. The SD configuration should not have anything to do with SPIFFS though

xcs011, 2021-04-28:
After searching around, neither grounding pin 0, or entering flash mode via onboard buttons works.

xcs011, 2021-04-28:
Trying to erase_flash with esptool cant connect either, I think something may be wrong with my board.

xcs011, 2021-04-28:
If I end up having to buy a new board, would you have any suggestions?

justcallmekoko, 2021-04-29:
Something simple like a ESP32 NodeMCU would be just fine.

justcallmekoko, 2021-06-02:
any progress?

xcs011, 2021-06-03:
The replacement board I ordered never came, im going to have to order another.

justcallmekoko, 2021-06-03:
Oh wow where did you order from?

xcs011, 2021-06-03:
I had ordered from PMD way, which had originally said it was shipping from Canada, then the tracking info showed China, and despite tracking saying it was delivered, it never came.

justcallmekoko, 2021-06-03:
What country are you in?

xcs011, 2021-06-03:
The United States

justcallmekoko, 2021-06-03:
Why not order from amazon?

xcs011, 2021-06-03:
It was an original d32, figured id go with it just to be safe, and still half the price.

justcallmekoko, 2021-06-07:
Let me know when you make some progress on this. I have been in contact with someone who managed to get this working with an ordinary ESP32 development board.

xcs011, 2021-07-31:
I know I havent posted in a while, I was busy with job searching, but I got a feather32 and will try again over the next few days.

xcs011, 2021-07-31:
I also realized this entire thing was my fault, I plugged it in upside down the first time and fried it because I wasn't paying attention.

xcs011, 2021-07-31:
I appear to have gone 1 step forward 2 steps back however, as now it just times out waiting for a packet header.

justcallmekoko, 2021-08-02:
Are you using the Arduino IDE for this or are you using esptool on command line?

xcs011, 2021-08-03:
I am using the arduino ide set up as in the guide.

justcallmekoko, 2021-08-12:
Can you provide a screenshot of the IDE output when you try to flash the firmware. Can you also send a picture of the ESP32 development board you are using?

xcs011, 2021-08-19:
I'm sorry I have been very busy job searching over the last few days and will re-open this when I have time.

closed 2021-08-19 by xcs011