justcallmekoko/ESP32Marauder #83: the code does not fit into the microcontroller memory

russo00, 2021-05-09:
Decided to carry out this wonderful project. I took the esp32-S Node MCU  and the display. The version 0.9.0 code does not fit into the flash memory of the microcontroller. What to do? Maybe there are older versions of the project with a smaller code size?

justcallmekoko, 2021-05-09:
What partition scheme are you choosing in the arduino IDE?

russo00, 2021-05-09:
does not compile to Node MCU ESP-32S board.
The sketch uses 1488117 bytes (113%) of device memory. Total 1310720 bytes available. Text section exceeds available space in board

russo00, 2021-05-09:
Partition Scheme is missing from the menu

justcallmekoko, 2021-05-09:
Select Lolin D32 as the board and select small SPIFFS as the partition scheme

GrandJohnny, 2021-05-24:
Hello there
ive changed the partition scheme as you said but now im getting something different:

`Sketch uses 1487181 bytes (75%) of program storage space. Maximum is 1966080 bytes.
Global variables use 108828 bytes (33%) of dynamic memory, leaving 218852 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port COM3
Connecting.......
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 8c:aa:b5:a2:91:ac
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content

`

btw what a cool project man keep it up!! :D

GrandJohnny, 2021-05-24:
its all fine when i disconnect esp32 from other stuff
i guess i screwed somewhere around wirings

justcallmekoko, 2021-05-25:
Which esp32 are you using?

GrandJohnny, 2021-05-25:
nodemcu 32s but i uploaded as a Lolin so it can fit in

justcallmekoko, 2021-05-25:
Have you already uploaded the SPIFFS image?

GrandJohnny, 2021-05-25:
yeah i uploaded them fine

GrandJohnny, 2021-05-26:
so i googled around and i think i got a solution
so first thing after i power the board with every single jumper connected is a rapid crash
then when i use an external power source for 5v and GND of my LCD again it goes crazy with a new crash because of SD_CS connected to GPIO12
so i realized on google that the problem maybe solved by connecting the SD_CS to some other GPIO but i dont where i should change it in the code and which GPIO would be better for it ?
and also i wonder marauder doesnt start successfully without SD_CS connected, i mean SD card and this pin is really needed or i am missing something else too?

justcallmekoko, 2021-05-26:
You should not need the SD card or the A32u4 connected. The marauder is designed to function without both. If you are still getting the same results without having the SD card connected, the issue is not GPIO12.

Have you tried another ESP32 module?

justcallmekoko, 2021-05-26:
It looks like there might be a solution to your problem [here](https://esp32.com/viewtopic.php?t=5703).
It sounds like the ESP32 is failing to find the bootloader at address 1000 if I understand the topic post I linked.

Read through it and try some of their suggestions to get it working again. Either that or try another ESP32 dev board.

GrandJohnny, 2021-05-30:
so i bought a new ESP32-WROOM but again after uploading it doesnt work and shows this :

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

do you think i maybe messed up with wiring?
because my 2.8 waveshare LCD pins were really different than what i seen in ic connections wiki

GrandJohnny, 2021-05-30:
so i tried to figure out which connection is messing this up
first i my LCD https://www.waveshare.com/wiki/2.8inch_TFT_Touch_Shield doesnt have these pins :
RST ( it has its button but i dont know how to use it )
T_CLK
T_DI
T_DO
T_CS ( it has TP_CS so i used it instead )
LED ( it has LCD_BL i thought it may be the same im using it)

but the problem occurs when i connect SD_CS to GPIO12
and also i dont know why im getting an empty screen without SD_CS with this output:

GrandJohnny, 2021-05-30:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Using TFT Shield
clearScreen()
SPI_FREQUENCY: 27000000
SPI_READ_FREQUENCY: 20000000
SPI_TOUCH_FREQUENCY: 2500000
clearScreen()

--------------------------------

         ESP32 Marauder      

            v0.9.1

       By: justcallmekoko

--------------------------------

Stopping BLE scan...
BLE Scan Stopped
Clearing BLE Results...
Deinitializing BT Controller...
RAM Free: 156932 bytes
Failed to mount SD Card
SD Card NOT Supported
RAM Free: 161404 bytes
RAM Free: 161404 bytes
RAM Free: 161404 bytes
RAM Free: 161308 bytes
IP5306 I2C Supported: false
RAM Free: 161136 bytes
Setting neopixel to black...
Checking for ESP8266...
[!] Setting ESP12 in run mode...
[!] Complete

Display string: 
Setup A32U4 Serial Interface
A32U4 Said: 0
initScrollValues()
setupScrollArea()
   tfa: 48
   bfa: 0
yStart: 48
Bulding buttons...
Displaying current menu...
clearScreen()
Got A32U4 Serial data
255

GrandJohnny, 2021-05-30:
is it because only ILI9241 is supported?

justcallmekoko, 2021-06-07:
@GrandJohnny What screen driver is used for the display you are using? The TFT_eSPI library supports multiple screen drivers. As for the SD_CS, I don't know why this would be preventing it from working properly. The Marauder functions properly without an SD card connected. It is designed to function with or with out it.

Are you still only getting an all white screen or are you actually getting something to display when you boot?

justcallmekoko, 2021-06-16:
Closing this issue as a solution to the original question was posted. @GrandJohnny if you are still having an issue with the screen, please post a separate issue.

closed 2021-06-16 by justcallmekoko