justcallmekoko/ESP32Marauder #86: The captured pcap file cannot be opened with wireshark

LuckyFishGeek, 2021-07-10:
[image]
[image]

justcallmekoko, 2021-07-11:
Can you please share the pcap file here?
[probe_1.zip](https://github.com/justcallmekoko/ESP32Marauder/files/6796124/probe_1.zip)
Try opening this pcap file in your wireshark. I captured this using the ESP32 Marauder

LuckyFishGeek, 2021-07-11:
> Can you please share the pcap file here?
> [probe_1.zip](https://github.com/justcallmekoko/ESP32Marauder/files/6796124/probe_1.zip)
> Try opening this pcap file in your wireshark. I captured this using the ESP32 Marauder

Thank you for your answer and help this is the file I captured.
[probe_0.zip](https://github.com/justcallmekoko/ESP32Marauder/files/6796126/probe_0.zip)

justcallmekoko, 2021-07-11:
I was also not able to open it. See if you can open pcap with less data captured by the marauder

LuckyFishGeek, 2021-07-11:
> I was also not able to open it. See if you can open pcap with less data captured by the marauder

ok   thanks   i try it

LuckyFishGeek, 2021-07-11:
Another question is that the screen display does not seem to support the hotspot name of UTF-8? I have a Chinese hot name to show that it is garbled.
[image]

justcallmekoko, 2021-07-11:
This is not an issue I encountered in my tests (which mean I have poor testing) so I never thought to implement a fix for this. I will have to figure out how to fix that

LuckyFishGeek, 2021-07-11:
> This is not an issue I encountered in my tests (which mean I have poor testing) so I never thought to implement a fix for this. I will have to figure out how to fix that

Thank you for your patience. I suspect that the language of other countries may also meet the problem of garbled, your work is very good, thank you!

justcallmekoko, 2021-07-11:
Thank you for the support. I will have to make sure the TFT library I use has multi-language support.

Let me know if a new pcap works

LuckyFishGeek, 2021-07-11:
> Thank you for the support. I will have to make sure the TFT library I use has multi-language support.
> 
> Let me know if a new pcap works

I have a bold guess, I can't upgrade the ProBe Request Sniff to support the specified MAC address tracking once the specified Mac will be displayed separately on the screen then ProBe Request Sniff changes TXT format, such as Mac ----- -Ap ------ Time

LuckyFishGeek, 2021-07-11:
> Thank you for the support. I will have to make sure the TFT library I use has multi-language support.
> 
> Let me know if a new pcap works

I have a further expectation of the feature ProBe Request Sniff. For example, save the pcap file in txt or log format, log format MAC -----AP ----time

You can specify the mac address detection, once the specified MAC address is detected will be displayed on the screen alone

In this case ProBe Request Sniff function is better

I just said a concept, but I can not code to build work haha hope you do not mind
Thanks