justcallmekoko/ESP32Marauder #88: Create SD Flasher firmware

justcallmekoko, 2021-07-19:
With the intention of releasing an ESP32 Marauder Kit. I want to make sure the process for building the kit is a simple as possible. Currently, the plan is to have the user physically assemble the kit, build the firmware from source, and flash the firmware using the Arduino IDE. 

With all of the libraries to be installed, library modifications to be made, and additional steps to be taken, I think it would be beneficial to the end user if there was a simple way to just upload the Marauder firmware using the actual precompiled bin file without having to use esptool. I myself am not familiar with the process of flashing the Marauder firmware using esptool from the command line.

The idea would be to install an SD Flasher firmware which would be built from source and uploaded using the Arduino IDE. This would hopefully mean that the end user would not to install any libraries and would only need to install the ESP32 boards from the board manager. Once the SD Flasher firmware is flashed to the ESP32, the end user would place the Marauder firmware bin on the root of the SD card, insert the SD card into the card reader of the Marauder, and power on the Marauder. The SD Flasher firmware would automatically detect the new firmware file and flash it to the ESP32.

This would be the process to follow in lieu of just using the esptool from CLI. This is also an option.

justcallmekoko, 2021-07-24:
Decided to make this OTA Flasher firmware so users would not be forced to buy an SD card.
a78d5b792cc22b2b308e3c3759c76efba3cbb69e

closed 2021-07-24 by justcallmekoko

tobozo, 2021-07-24:
some microSD cards are really cheap, those are sub-dollar on aliexpress

[image]