justcallmekoko/ESP32Marauder #95: Failed to Initialize SD Card

sfernandez39, 2021-07-29:
Serial Monitor Output

18:19:09.253 -> ESP-IDF version is: v4.4-dev-2030-gd93887f9f-dirty
18:19:09.577 -> Using TFT Shield
18:19:09.577 -> clearScreen()
18:19:09.624 -> SPI_FREQUENCY: 27000000
18:19:09.624 -> SPI_READ_FREQUENCY: 20000000
18:19:09.624 -> SPI_TOUCH_FREQUENCY: 2500000
18:19:09.624 -> ===============
18:19:09.624 -> JPEG image info
18:19:09.624 -> ===============
18:19:09.624 -> Width      :240
18:19:09.624 -> Height     :320
18:19:09.624 -> Components :1
18:19:09.624 -> MCU / row  :15
18:19:09.624 -> MCU / col  :20
18:19:09.624 -> Scan type  :4
18:19:09.624 -> MCU width  :16
18:19:09.624 -> MCU height :16
18:19:09.624 -> ===============
18:19:09.624 -> 
18:19:11.767 -> clearScreen()
18:19:13.814 -> 
18:19:13.814 -> 
18:19:13.814 -> --------------------------------
18:19:13.814 -> 
18:19:13.814 ->          ESP32 Marauder      
18:19:13.814 -> 
18:19:13.814 ->             v0.9.3
18:19:13.814 -> 
18:19:13.814 ->        By: justcallmekoko
18:19:13.814 -> 
18:19:13.814 -> --------------------------------
18:19:13.814 -> 
18:19:13.814 -> 
18:19:14.092 -> Stopping BLE scan...
18:19:14.092 -> BLE Scan Stopped
18:19:14.092 -> Clearing BLE Results...
18:19:14.092 -> Deinitializing BT Controller...
18:19:14.092 -> RAM Free: 119748 bytes
18:19:14.695 -> Failed to mount SD Card
18:19:14.695 -> SD Card NOT Supported
18:19:14.695 -> RAM Free: 124212 bytes
18:19:14.695 -> RAM Free: 124212 bytes
18:19:14.741 -> RAM Free: 124212 bytes
18:19:14.741 -> RAM Free: 124116 bytes
18:19:14.741 -> IP5306 I2C Supported: true
18:19:14.741 -> RAM Free: 123952 bytes
18:19:14.741 -> Setting neopixel to black...
18:19:17.332 -> initScrollValues()
18:19:17.332 -> setupScrollArea()
18:19:17.332 ->    tfa: 48
18:19:17.332 ->    bfa: 0
18:19:17.332 -> yStart: 48
18:19:17.612 -> Bulding buttons...
18:19:17.612 -> Displaying current menu...
18:19:17.612 -> clearScreen()

Steps to Recreate
[OG_ESP32_Marauder_-_Failed_to_Initiliaze_SD_Card.txt](https://github.com/justcallmekoko/ESP32Marauder/files/6903685/OG_ESP32_Marauder_-_Failed_to_Initiliaze_SD_Card.txt)
/Troubleshoot

1. Connect the charger while the marauder is on, the SD cards fails to initialize when it reboots. 
2. Unplugging the Marauder, turning it off, removing the SD card, reinserting it and turning the Marauder ON seem to clear it. 
3. If I connect the charge with the Marauder off, SD card initializes fine when the Marauder starts.

justcallmekoko, 2021-07-30:
I have a feeling this might have something to do with the SD card being powered (when just by battery) and initialized via SPI by the ESP32. When the USB is plugged in, power is briefly cut as the IP5306 transitions from battery power to USB power. I don't know if this causes the SD card to restart, but it definitely causes the ESP32 to restart. I suspect it leaves the SD card in a weird state if it does not actually power off. Maybe the SD card is in an "initialized" state and the Marauder at its second boot is trying to reinitialized it and the SD card is like, "bro I'm there".

justcallmekoko, 2022-09-08:
Fixed in release v0.9.15
https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.9.15

closed 2022-09-08 by justcallmekoko