justcallmekoko/ESP32Marauder #98: analog battery measurement

superjasiek, 2021-09-09:
Hi,
Thanks for the cool project and the previous advice. Everything compiled as it should be. I have one more question - where and how to modify the code to compile it for old hardware? My design is based on the NodeMcu32-s and the 2.8 'touchscreen, when I compile the current version, the touchscreen works the opposite. I made an update to "old hardware" (with precompiled bin file) and everything is fine but I want to turn on analog battery measurement ("#define BATTERY_ANALOG_ON"  1).
Thanks in advance for the tips.
One more thing - where do I find Evil Twin and Captive Portal attacks?

justcallmekoko, 2021-09-09:
> My design is based on the NodeMcu32-s and the 2.8 'touchscreen, when I compile the current version, the touchscreen works the opposite. I made an update to "old hardware" (with precompiled bin file) [...]

If you are using a 2.8" touch screen PCB like the one on the OG Marauder, you have to make sure `#define TFT_SHIELD` is uncommented and `#define TFT_DIY` and `#define KIT` are commented. If you do not do this, the screen touch coordinated will be inverted.

> One more thing - where do I find Evil Twin and Captive Portal attacks?

I have not developed these capabilities yet. Right now they are only plans.

justcallmekoko, 2021-09-10:
Glad you got it working. Don't worry about the neopixel. It isn't actually used for anything. I was going to use it as a status indicator but when the ESP32 boots and initialized the neopixel library, it causes the neopixel to flash once really brightly before turning off and it annoyed me. I removed the hardware feature from all new generations of Marauder so really that source code for the neopixel is a phantom limb.

closed 2021-09-10 by justcallmekoko