justcallmekoko/ESP32Marauder #102: Suggestion regarding: Wi-Fi Behavior

6q6, 2021-10-17:
On the current firmware 0.9.2 there are multiple optimizations possibilities regarding the behavior of Wi-Fi during:
-DeauthFloods:
When Deauthentication Floods are currently executed an ESP open SSID broadcast is started. If possible this should be disabled (or hidden if it breaks functionality) during ANY Wi-Fi operation since it impacts the esp32marauders covert effectiveness. 

With this enabled, Monitoring tools could enumerate the serial of the ESP32 chip and it's broadcasting MAC address. 

-Upgrading Firmware:
After the firmware upgrade via WebGUI, the devices restart and keep broadcasting "marauderOTA", this leads to a full enumeration of the device used, if not manually shut down, by starting and closing the updater. 

-Wi-Fi Mac randomization at startup:
Speaks for itself, more discretion.

Crsarmv7l, 2022-04-14:
AP mode is not necessary to inject raw packets, you can go through and change his code to support sending from STA if you want. Then you wont have an AP

maxkarelov, 2023-01-19:
@Crsarmv7l could you please show the code part related to AP mode to disable it ?

Crsarmv7l, 2023-01-19:
There is a caveat to what I said. If your esp32 doesn't have a screen and controls, then you *need* ap mode to interface with marauder.

It would also be a rather extensive re-write to get away from AP mode on marauder. I ended up completely implementing my own solution to fit in with the platform I was working with. I did both deauth and 6 char SSID beacon spam based on koko's work.

Mine is laid out here https://github.com/Crsarmv7l/My-TTGO-Watch-SFOS/blob/Nimble-android/src/gui/mainbar/setup_tile/wlan_settings/wlan_settings.cpp

maxkarelov, 2023-01-21:
@Crsarmv7l I am using flipper zero companion app to interface esp32 via uart.

Looking for ways to fork and disable it in firmware and upload that custom to esp32