justcallmekoko/ESP32Marauder #105: Can this project detect non-discoverable devices?

WillNilges, 2022-01-30:
Howdy, I'm working on a similar project, and I'm wondering if the Marauder can detect non-advertised/non-discoverable devices?

Crsarmv7l, 2022-03-15:
Like hidden accesspoints? It should be able to. If you look at his code, he is using promiscuous mode, so gathering all packets and filtering them. Should be able to detect a hidden network, and even if it isn't implemented it shouldnt be hard.

WillNilges, 2022-03-15:
Sorry, should've been more clear that I was asking about Bluetooth devices that weren't in advertising mode. iPhones connected to Apple Watches and stuff like that. The TL;DR of the issue I was seeing is that typically, in order for a device to show up like that, _it_ needs to respond to _your_ advertising packet. I don't know if there's any way around that, but I was hoping there'd be one here.

Crsarmv7l, 2022-04-17:
I would expect not. I am certainly not an expert on bt, (although I know a fair bit about bt characteristics, descriptors, etc), but from my understanding, advertising mode is needed to initially pair so each device knows the address of the other, from there it can be disabled because you already have the addresses and they can just talk directly. 

Without the visibility you would need to try and bruteforce the address, and even then the bonding process means the device only responds to the address it is paired/bonded with so you would need to figure out both device addresses, and spoof the address of the sender on the bt module of the esp32 (which I am not sure can be done).

Crsarmv7l, 2022-04-17:
I did find this. Notably it seems to only work with BT Classic and not BLE
https://github.com/Matheus-Garbelini/esp32_bluetooth_classic_sniffer