justcallmekoko/ESP32Marauder #107: "collect2.exe: error: ld returned 1 exit status" Need help please

Irish1982, 2022-03-06:
I have beenfighting with this thing for like 6 months but i was using devkitv1 doit esp32 and now im using a adafruit huzzah32 feather. 
I am getting this error toward the end of compiling 
[esp32marauder.txt](https://github.com/justcallmekoko/ESP32Marauder/files/8193091/esp32marauder.txt)
I am using the Adafruit Huzzah32 feather with a Seeedstudio 2.8 TFT touchshield V2.1 with ili9341, so could anyone give me any advise on the file i included cause i am stumped lol any advise would be awesome thank you so much

justcallmekoko, 2022-03-06:
It looks like you did not setup your IDE as described [here](https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup#these-next-steps-only-apply-if-you-plan-to-build-the-full-esp32-marauder-firmware-from-source). You need to make sure you allow multiple definitions of functions. This is how the sanity check imposed by espressif is bypassed to allow deauthentication frames.

justcallmekoko, 2022-03-06:
#97

closed 2022-03-06 by justcallmekoko

Irish1982, 2022-10-11:
I have gotten the marauder running on my old esp32 board. However I've got
a small issue, how do I get the touch screen working. I have it running on
a 2.8 tft touchshield v2.1 seeedstudio. It has A0,A1,A2,A3 which have
-y,-x,+y,+x pins. Do I use it to get the touch to work. I was under the
impression the spi pins would work but, its most definitely not working and
if I use those pins what library would I have to add to the current library
to make them work. Any help is much appreciated.