justcallmekoko/ESP32Marauder #114: support for m5 fire stack

gtgre, 2022-05-09:
can you please relese documentation and code for the m5 fire stack?

201nasko, 2022-05-09:
yes agreed!!! and or m5stack core 2 (the one im trying to port it too)

justcallmekoko, 2022-05-10:
I am currently developing for compatibility with one of the new pieces of hardware I've designed so I will probably not be getting around to this anytime soon but you are more than welcome to take a crack at it yourself especially if you can get it working with m5stack faces

r00sta, 2022-05-29:
@201nasko have you had much luck? I'm looking to do the same...

201nasko, 2022-05-29:
@r00sta i've managed to get it on my m5stack core 2 but have not yet managed to get it on my adafruit pyportal titano... ill post documenting how to do it a little later today!

rexido1, 2022-06-12:
Who knows why this error occurs: exit status 1
'updateMenu' was not declared in this scope ?

#ifdef HAS_SCREEN
        menu_function_obj.changeMenu(updateMenu);
#endif

justcallmekoko, 2022-06-13:
@rexido1 when was the last time you pulled from the repo?

rexido1, 2022-06-13:
> @rexido1 when was the last time you pulled from the repo?

one day ago

LuckyFishGeek, 2022-11-22:
@201nasko  Can you share how to make m5 fire stack? thank you very much

justcallmekoko, 2022-12-23:
Support for M5 stack will have to be user-based. It is possible to build M5 stack compatible firmware from source. You would need to add configurations for button input and ensure the proper screen configuration is selected upon compilation.

closed 2022-12-23 by justcallmekoko

7h30th3r0n3, 2023-02-01:
> @r00sta i've managed to get it on my m5stack core 2 but have not yet managed to get it on my adafruit pyportal titano... ill post documenting how to do it a little later today!

any news about m5stack core 2 documentation ? trying to make the same too 😋

Aro2142, 2024-02-20:
I did it, but....

https://github.com/justcallmekoko/ESP32Marauder/assets/48096137/2a52c2b5-a413-4d3f-ac8c-fa9d086c3cbf

Display ILI9342C.  Probably you need to use a different driver, not for ILI9341
😁

Aro2142, 2024-02-20:
> Display ILI9342C. Probably need to use a different driver, not for ILI9341 😁

It helped, but not much

https://github.com/justcallmekoko/ESP32Marauder/assets/48096137/7848130f-d442-46dc-a949-460fae45ec79

Aro2142, 2024-02-20:
Any ideas?)

7h30th3r0n3, 2024-02-20:
> Any ideas?)

Well , on my side, I tried quite a few things, but nothing seems to work so I decided to create my own firmware

Aro2142, 2024-02-23:
> > Any ideas?)
> 
> Well , on my side, I tried quite a few things, but nothing seems to work so I decided to create my own firmware

Yeah that's why I made this shit, but it works quite well.  Now 
[image]
need to assemble the Python code to make it all look nice

Aro2142, 2024-02-23:
M5 Stack will control esp32 like flipper zero lol

Aro2142, 2024-02-23:
Ahhhh, it doesn’t work stably, it seems that the M5stack’s power supply is not enough, even pcram stops working...

Aro2142, 2024-02-25:
I decided to make a huge crutch for M5Stack and will soon create a GIT repository for my project. It's a terribly dumb way to get Marauder to work on the M5, but it's better than nothing.
The project name will be "M5Stack marauder helper" XD

7h30th3r0n3, 2024-02-25:
Maybe check the Evil-M5core2 firmware, better than nothing haha 😜

Aro2142, 2024-02-25:
> Maybe check the Evil-M5core2 firmware, better than nothing haha 😜

I have already checked Evil-M5core2.  I hope that your project has a great future ahead, but unfortunately there is no function for sending Deauther frames yet.  If I could, I wouldn't waste time on this if I knew how to send these damn frames, but my brain isn't strong enough for that😁.

7h30th3r0n3, 2024-02-26:
Deauth should be implemented in near future, I got lot of demand for this functionality