justcallmekoko/ESP32Marauder #115: issues trying to get firmware on Adafruit Pyportal Titano & M5Stack Core 2

201nasko, 2022-05-09:
both have all the necessary parts and chips cant manage to get them to work though any help?

justcallmekoko, 2022-05-10:
What have you tried?
What is the expected result?
What is the actual result?

201nasko, 2022-05-10:
so i follow the directions exactly as they say to but i of course change the setting according to the board im using (pyportal titano) and when i hit upload it gives me "error: esp_wifi.h no such file or directory" and i cant find the file to include in it... this is the only problem i have on the pyortal.. i'm going to try on my m5stack and see if i can get a different result

justcallmekoko, 2022-05-10:
Please include the whole compiler error when you get a chance

201nasko, 2022-05-10:
esp32_marauder:10:10: fatal error: esp_wifi.h: No such file or directory
   10 | #include "esp_wifi.h"
      |          ^~~~~~~~~~~~
compilation terminated.
exit status 1
esp_wifi.h: No such file or directory