justcallmekoko/ESP32Marauder #127: Flipper Zero Wifi board dead after firmware installation.

bitcoin42, 2022-06-13:
After installing the firmware the board is dead. Nothing works. Tried to reset or reboot button, nothing works anymore after the last step of the installation. 

What to do?

justcallmekoko, 2022-06-13:
Describe the exact steps you took to install the firmware and troubleshoot the issue. Provide me a sample of serial output from the wifi dev board

bitcoin42, 2022-06-13:
I flowed exactly these steps:
https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup

then these:
https://github.com/justcallmekoko/ESP32Marauder/wiki/installing-firmware-via-ota

After step 15. the board was dead. Nothing makes it react, and I can't read anything from it. It doesn't get recognized and the reboot or reset button doesn't make anything.

I don't know how to give you a sample.

BIZZLE947, 2022-06-20:
Jumping on this same issue...

After following your steps here is the serial output I received via arduino IDE:
[serial output.txt](https://github.com/justcallmekoko/ESP32Marauder/files/8940382/serial.output.txt)

Appreciate any help, cheers

justcallmekoko, 2022-06-20:
@BIZZLE947 That looks like expected output. You should be able to use it now

justcallmekoko, 2022-07-06:
Any updates on your end?

Patocoh, 2022-07-14:
having the same issue, when I added to the Arduino with the installation I saw the following info,

Sketch uses 738606 bytes (37%) of program storage space. Maximum is 1966080 bytes.
Global variables use 32668 bytes (9%) of dynamic memory, leaving 295012 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.1
Serial port COM12
Connecting...
Chip is ESP32-S2
Features: WiFi
Crystal is 40MHz
MAC: 12:88:24:23:00:30
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x000c4fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 542.1 kbit/s)...
Hash of data verified.
Compressed 13648 bytes to 9382...
Writing at 0x00001000... (20 %)
Writing at 0x00001c1f... (40 %)
Writing at 0x0000276e... (60 %)
Writing at 0x00003294... (80 %)
Writing at 0x00003e95... (100 %)
Wrote 13648 bytes (9382 compressed) at 0x00001000 in 0.3 seconds (effective 328.5 kbit/s)...
Hash of data verified.
Compressed 738992 bytes to 502858...
Writing at 0x00010000... (0 %)
Writing at 0x00010953... (0 %)
--snipped--
Writing at 0x000c3f72... (100 %)
Wrote 738992 bytes (502858 compressed) at 0x00010000 in 8.1 seconds (effective 732.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 318.5 kbit/s)...
Hash of data verified.

Leaving...
ERROR: ESP32-S2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this error, set --after option to 'no_reset'.
To suppress this error, set --after option to 'no_reset'.

justcallmekoko, 2022-07-15:
@Patocoh This also looks like expected output. Now you would just hit the physical reset button on the board

SherifNagy, 2022-07-29:
I thought I had the same issue, I was fooled by the LED not blinking and flashed the device few times! but it works as described :)

- followed the video / guide instruction
- once this is done, the device did reboot
- plug it into the flipper zero
- connect the flipper zero to PC
- open minicon on -b 115200
- flipper zero GPIO speed is set to host
- press reset button on the module
- and I got a CLI !

hope this helps!

Patocoh, 2022-07-29:
Thanks, @justcallmekoko @SherifNagy it worked. I believe it was missing to set the clock to 115200 and then reset!

closed 2022-08-02 by justcallmekoko