justcallmekoko/ESP32Marauder #128: Flipper zero ESP32-s2 attack error

vaklama, 2022-07-04:
Hi,

Scan works but when I try to attack I get this error message:

Starting (xy) attack. Stop with stopscan
> E (149171) wifi:sched is null, if=1 tid=7 flags=6402
E (149171) wifi:trc.c rcReachRetryLimit 1733

After the error message, only a reset helps

esp32_marauder_v0_9_9_20220628_flipper
flipper fw: 0.61.1

justcallmekoko, 2022-07-05:
Yea I've been googling my ass off about this. I will need help finding a solution

Bungeetaco, 2022-07-06:
Same thing here, tried both a deauth attack on my home network, and a rickroll attack, and both return the same error code:

> E (163317) wifi:sched is null, if=1 tid=7 flags=6402
E (163317) wifi:trc.c rcReachRetryLimit 1733

justcallmekoko, 2022-07-06:
This issue seems to affect ESP32-S2 modules but not standard ESP32 modules. Can confirm on the WROOM and WROVER modules.

deft01, 2022-07-07:
Same here with all attacks :/

vaklama, 2022-07-07:
> This issue seems to affect ESP32-S2 modules but not standard ESP32 modules. Can confirm on the WROOM and WROVER modules.

Wrover modul

FroggMaster, 2022-07-10:
I can confirm that I'm getting the same errors with all attacks as well. 
It happens within a few seconds of starting the attack.

FroggMaster, 2022-07-10:
I don't know if any of this is of any benefit you to but I found this discussion online that seems to have a similar error. 
https://github.com/espressif/esp-idf/issues/6883

justcallmekoko, 2022-07-18:
Fixed in https://github.com/justcallmekoko/ESP32Marauder/commit/cbcfa1033077d8ea972c5bcfa408a36182b97b1d

closed 2022-07-18 by justcallmekoko