justcallmekoko/ESP32Marauder #131: flipper zero issues

CTRL-ALT-UP, 2022-07-26:
Hi , thanks for the amazing build.

I'm having issues with the v1 WiFi dev board not retaining the firmware. esp32_marauder_v0_9_10_20220718_flipper.bin

I can get it to upload the firmware via the OTA web interface and get it to fully function on the flipper  but if I reboot the dev-board either by the switch or CLI it dumps the firmware. If the flipper is turned off or the the dev-board removed from the gpio it also loses the firmware and I have to upload it again via the OTA web interface. Here is a CLI dump after the last reboot.

22:11:07.959 ESP‑ROM:esp32s2‑rc4‑20191025
22:11:07.962 Build:Oct 25 2019
22:11:07.963 rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
22:11:07.968 SPIWP:0xee
22:11:07.969 mode:DIO, clock div:1
22:11:07.970 load:0x3ffe6100,len:0x524
22:11:07.973 load:0x4004c000,len:0xa70
22:11:07.975 load:0x40050000,len:0x292c
22:11:07.978 SHA‑256 comparison failed:
22:11:07.980 Calculated: 8dce998efc3eb6dc43f3799a489dbcfbd5ada9828beadf71017c05c4a8d32bc3
22:11:07.987 Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
22:11:07.993 Attempting to boot anyway...
22:11:07.996 entry 0x4004c18c
22:11:08.247 
22:11:08.248 
22:11:08.249 Marauder OTA Update
22:11:08.250 
22:11:08.250 Configuring update server...
22:11:08.251 Initializing WiFi...
22:11:08.289 Starting softAP...
22:11:08.296 
22:11:08.297 Displaying settings to TFT...
22:11:08.300 IP address: 192.168.4.1
22:11:08.302 Setting server behavior...
22:11:08.304 Finished setting server behavior
22:11:08.307 Starting server...
22:11:08.309 Completed update server setup
22:11:08.311 
22:11:08.311 ‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑‑
22:11:08.315 
22:11:08.315 SSID: MarauderOTA
22:11:08.316 Network Password: justcallmekoko
22:11:08.319 Username: admin
22:11:08.321 Password: admin

Any help would be appreciated

justcallmekoko, 2022-07-27:
What partition scheme did you select when you uploaded the OTA firmware

ThornWalli, 2022-07-27:
Have the same problem, I selected 
[image]

CTRL-ALT-UP, 2022-07-27:
I was the same

Regards
Keith

[image]<https://user-images.githubusercontent.com/8287751/181363204-53f30c42-a651-420f-8ad8-34e6212e2c37.png>

ThornWalli, 2022-07-27:
Solved it for me...

Have not followed the given previous version `2.0.0-rc1` for `esp32`.

Now I can plug in and out.

CTRL-ALT-UP, 2022-07-27:
Thanks, Thats solved it for me too. All i can say about my self is numpty.

if you're not scottish.

[Numpty](https://www.urbandictionary.com/define.php?term=Numpty)

Scottish usage:
a) Someone who (sometimes unwittingly) by speech or action demonstrates a lack of knowledge or [misconception](https://www.urbandictionary.com/define.php?term=misconception) of a particular subject or situation to the amusement of others.

justcallmekoko, 2022-07-28:
lol ok then make sure you install `2.0.0-rc1`

closed 2022-07-28 by justcallmekoko