justcallmekoko/ESP32Marauder #133: A fatal error occurred: Failed to connect to ESP32-S2: Timed out waiting for packet header

nastea1, 2022-07-28:
I assume I'm missing the driver on the com port, it says "none on com4" or "none on com7" even though I pick the right board, etc. I've got the 2.0.0rc drivers loaded, so I"m assuming my stupid windows computer isn't loading the com drivers correctly. It lists them as microsoft drivers which I know is wrong.

A fatal error occurred: Failed to connect to ESP32-S2: Timed out waiting for packet header
Sketch uses 751246 bytes (38%) of program storage space. Maximum is 1966080 bytes.
Global variables use 32500 bytes (9%) of dynamic memory, leaving 295180 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.1
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32-S2: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32-S2: Timed out waiting for packet header

justcallmekoko, 2022-07-29:
Have you tried holding the boot button, hitting the reset button, then releasing the boot button?

nastea1, 2022-07-29:
That was it!

closed 2022-07-29 by nastea1