justcallmekoko/ESP32Marauder #134: Unable to upload Marauder on Flipper Zero: "OSError: [Errno 71] Protocol error"

otanim, 2022-07-30:
**OS:** Ubuntu 20.04
**Kernel:** 5.14.0-1045-oem
**Arduino IDE:** 1.8.19
**esp32:** 2.0.0-rc1 / 2.0.4 (tested on both)
**USB port:** USB-C -> USB-C
**Verbose log (upload only, the compile log was too long to post):**
```
Sketch uses 765234 bytes (38%) of program storage space. Maximum is 1966080 bytes.
Global variables use 33020 bytes (10%) of dynamic memory, leaving 294660 bytes for local variables. Maximum is 327680 bytes.
python /home/otanim/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py --chip esp32s2 --port /dev/ttyACM0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /home/otanim/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/partitions/boot_app0.bin 0x1000 /tmp/arduino_build_887905/My_Blink.ino.bootloader.bin 0x10000 /tmp/arduino_build_887905/My_Blink.ino.bin 0x8000 /tmp/arduino_build_887905/My_Blink.ino.partitions.bin 
esptool.py v3.1
Serial port /dev/ttyACM0
Traceback (most recent call last):
  File "/home/otanim/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 4582, in <module>
    _main()
  File "/home/otanim/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 4575, in _main
    main()
  File "/home/otanim/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 4076, in main
    before=args.before)
  File "/home/otanim/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 121, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/home/otanim/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 629, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False, usb_jtag_serial=usb_jtag_serial)
  File "/home/otanim/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 590, in _connect_attempt
    self.bootloader_reset(esp32r0_delay, usb_jtag_serial)
  File "/home/otanim/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 553, in bootloader_reset
    self._setDTR(False)  # IO0=HIGH
  File "/home/otanim/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 499, in _setDTR
    self._port.setDTR(state)
  File "/usr/local/lib/python2.7/dist-packages/serial/serialutil.py", line 603, in setDTR
    self.dtr = value
  File "/usr/local/lib/python2.7/dist-packages/serial/serialutil.py", line 473, in dtr
    self._update_dtr_state()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 715, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_DTR_str)
IOError: [Errno 71] Protocol error
IOError: [Errno 71] Protocol error

Connecting...
```

otanim, 2022-07-30:
Tested on Windows 10 OS, worked as instructed, no errors.

justcallmekoko, 2022-08-02:
Closing this as this sounds like an OS configuration issue

closed 2022-08-02 by justcallmekoko