justcallmekoko/ESP32Marauder #135: Issue with compiling for Arduino IDE

cookins, 2022-08-01:
hey there.
i downloaded last version and cant compile it in arduino IDE. if anybody cant give me hint, why its not compiling.. I building for DOIT esp32 devkit v1. Here logs:

```
C:\Users\cooki\Desktop\engi\ESP\esp32\ESP32Marauder-master\esp32_marauder\WiFiScan.cpp: In member function 'void WiFiScan::RunSetup()':
WiFiScan.cpp:138:19: error: 'setScanFilterMode' is not a member of 'NimBLEDevice'
     NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE);
                   ^~~~~~~~~~~~~~~~~
WiFiScan.cpp:139:19: error: 'setScanDuplicateCacheSize' is not a member of 'NimBLEDevice'
     NimBLEDevice::setScanDuplicateCacheSize(200);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\cooki\Desktop\engi\ESP\esp32\ESP32Marauder-master\esp32_marauder\WiFiScan.cpp: In member function 'void WiFiScan::RunBluetoothScan(uint8_t, uint16_t)':
WiFiScan.cpp:1071:19: error: 'setScanFilterMode' is not a member of 'NimBLEDevice'
     NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE);
                   ^~~~~~~~~~~~~~~~~
WiFiScan.cpp:1072:19: error: 'setScanDuplicateCacheSize' is not a member of 'NimBLEDevice'
     NimBLEDevice::setScanDuplicateCacheSize(200);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
WiFiScan.cpp:1111:15: error: 'class NimBLEScan' has no member named 'setMaxResults'; did you mean 'getResults'?
     pBLEScan->setMaxResults(0);
               ^~~~~~~~~~~~~
               getResults
Multiple libraries were found for "SD.h"
 Used: C:\Users\cooki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\SD
 Not used: D:\Program Files\Arduino\libraries\SD
 Not used: C:\Users\cooki\Documents\Arduino\libraries\SD
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\cooki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\WiFi
 Not used: D:\Program Files\Arduino\libraries\WiFi
exit status 1
'setScanFilterMode' is not a member of 'NimBLEDevice'

```

justcallmekoko, 2022-08-01:
This looks like you may have installed the wrong version of NimBLE

cookins, 2022-08-01:
> This looks like you may have installed the wrong version of NimBLE

ok, indeed. i updated all libs i saw in main file, but there no that lib included, or its included with another lib or file. anyway, thanks, i updated from 1.1 to 1.4. another small question - 
```
#define MARAUDER_MINI
  //#define MARAUDER_V4
  //#define MARAUDER_V6
  //#define MARAUDER_KIT
  //#define GENERIC_ESP32
  //#define MARAUDER_FLIPPER
```
i got DOIT esp32 devkit v1 board. if that definition correct - MARAUDER_MINI? Or should i change to GENERIC_ESP32?

cookins, 2022-08-02:
UP

justcallmekoko, 2022-08-02:
Can you provide the full error log from the Arduino IDE when it fails to compile?

cookins, 2022-08-02:
> Can you provide the full error log from the Arduino IDE when it fails to compile?

its very long to paste here. so part 1 https://pastebin.com/ucDxPJ5P and part 2 https://pastebin.com/MdG6pCWh

cookins, 2022-08-02:
i tried to compile with VSC. seems he now only complain about ` 'FreeMono9pt7b' was not declared in this scope `. and some warnings..

cookins, 2022-08-06:
UP

pfxr, 2023-05-29:
Any update on this? I am also having trouble compiling for DOIT Esp32 devkit v1. Did you manage to compile this?

cookins, 2023-05-30:
> Any update on this? I am also having trouble compiling for DOIT Esp32 devkit v1. Did you manage to compile this?

No.