justcallmekoko/ESP32Marauder #136: Easier and faster full flash for FlipperZero WiFi DevBoard

improving-rigmarole, 2022-08-01:
so I became tired real quick of having to flash through ArduinoIDE then OTA via WiFi each time I wanted to switch between FlipperBlackmagic and Marauder.

@UberGuidoZ [mentions](https://github.com/UberGuidoZ/Flipper/tree/main/Wifi_DevBoard) Frogg's script made for their [ESP32-Wi-Fi-Penetration-Tool](https://github.com/FroggMaster/ESP32-Wi-Fi-Penetration-Tool).

I just had to dump the *partition-table* and `bootloader` from the DevBoard once the Marauder installed, and replace them in the FW folder. I checked if those were identical between [v0.9.10](https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.9.10) and [v0.9.11](https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.9.11) and it's the case (but I think they're actually both just fixed binary outputs from the Arduino OTA script so I guess they won't need to be changed often).

It's also possible to tweak the script to make it use the firmware passed as argument (cli or drag'n'drop over .bat), for flexibility and ease of use.

UberGuidoZ, 2022-08-13:
I have tested the mentioned script, as well as some colleagues, and everyone loves it. Backup of existing firmware it included though giving issues. Otherwise, highly recommend checking this out.

justcallmekoko, 2022-08-29:
Implemented in Flipper Zero docs of wiki

UberGuidoZ, 2022-09-04:
Current version (with some improvements) is here: https://github.com/UberGuidoZ/Flipper/tree/main/Wifi_DevBoard/FZ_Marauder_Flasher