justcallmekoko/ESP32Marauder #141: Captured pcap file is empty

Louren, 2022-08-08:
Hi,

My captured PCAP files on a Flipper zero with the SD board are empty. All files (eg from scanap, sniffpmkid) are the same, examples attached.

I believe the issue is different then #86, since their files are >1kb, mine consistently 24 bytes. Also this is on a Flipper.

The console output reveals no issues, for example on first run:
```23:25:08.516 -> /SCRIPTS does not exist. Creating...
23:25:08.654 -> /SCRIPTS created
23:25:08.654 -> CLI Ready
```
The folder is also there when I read SD from mac.

When sniffing, the output is:
```23:08:19.517 -> > #channel -s 11
23:08:22.806 -> Set channel: 11
23:08:22.806 -> > #sniffpmkid -d
23:08:31.910 -> Starting PMKID sniff with deauthentication on channel 11. Stop with stopscan
23:08:31.910 -> /eapol_0.pcap
23:08:32.412 -> > Received EAPOL:
23:08:34.987 -> Received EAPOL:
23:08:34.987 -> Received EAPOL:
23:08:34.987 -> Received EAPOL:
23:08:34.987 -> Received EAPOL:
23:08:35.239 -> Received EAPOL:
```
This file is, again, succesfully created. But empty (no packets in wireshark).

[pcaps.zip](https://github.com/justcallmekoko/ESP32Marauder/files/9285177/pcaps.zip)

Thanks for any suggestions!

galactricks, 2022-08-14:
after soldering the micro SD support I see the same problem, the sound files are well written, however they are empty

Louren, 2022-08-28:
So update: I replaced the SD board (because the click meganism broke..): same issue.

No output that indicates any error. 

I'll replace the firmware with a fresh download from the github releases, will report back after.

I have tried different types of formatting SD, to no avail.

Any suggestions are welcome!

justcallmekoko, 2022-08-30:
I am working to get this fixed

Louren, 2022-08-30:
Cool, let me know if there's anything I can test or provide to help!

closed 2022-08-30 by justcallmekoko

BJamin99, 2022-08-30:
Thanks for this! I can confirm that 0.9.14 fixed the issue for me.