justcallmekoko/ESP32Marauder #142: FlipperZero wifi board doesn't respond to input

rwegrzyniak, 2022-08-09:
I've installed marauder software on FliiperZero wifi board via OTA. Unfotunately I cannot send anything to board. I've tried with custom FlipperOS https://github.com/RogueMaster/flipperzero-firmware-wPlugins and with USB-serial bridge with baud 115200. In both cases when I press boot button on the board, board responds with:

`ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x1 (DOWNLOAD(USB/UART0/1/SPI))
waiting for download`

then when I press restart I've got:

`
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASHSPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x498
load:0x4004c000,len:0xa74
load:0x40050000,len:0x25ec
entry 0x4004c194
ESP-IDF version is: v4.4-dev-2313-gc69f0ec32
Failed to mount SD Card
SD Card NOT Supported
CLI Ready`

then it prints logo and prompt but when I try to type `help` Flipper GPIO app shows that Bytes was transmited to Pin 13 but board doesn't respond.

I've tried reinstalling marauder on board but it didn't work.

rwegrzyniak, 2022-08-09:
I've tried Marauder companion app from RogueMaster Flipper firmware and on my Arch Linux PC  I've used `screen` to access serial connection  Flipper USB-UART bridge

justcallmekoko, 2022-08-09:
Is the output the same while using the companion app? It doesn't respond to input?

rwegrzyniak, 2022-08-09:
Yes, both options gives me same output. Last element in output always is `>` sign and nothing happens when I try to send any command.

justcallmekoko, 2022-08-09:
Does the blue light on the dev board turn on when you start a scan like with `scanap`?

rwegrzyniak, 2022-08-09:
no

tonyarris, 2022-08-10:
I have the same issue. Pressing down lightly on the WiFi board itself solves it for me. It's not very convenient to have to hold it down though.

rwegrzyniak, 2022-08-11:
I've tried few things with serial connection and if command `scanap` was pushed very soon after reboot ( pressing boot and after that restart) it's scanning for a couple of seconds - blue light appears and output is in terminal but after couple of seconds it's stops responding again. Also as @tonyarris mentioned if I keep boot button pressed then everything seems to work fine.

rwegrzyniak, 2022-08-11:
> Edit for clarity: I mean physically pushing the board itself into the GPIO ports to get a solid connection with the Flipper.

Oh I see, that was some misunderstanding.

In my case pushing board doesn't work, but I've checked couple times and keeping boot button pushed works thus I'm not sure if our cases are related.

hyperglitch, 2022-08-13:
I have the same issue. It looks like the board freezes just after printing the ">" prompt. I tried with usb-serial adapter and with flipper (both in uart bridge and with RM marauder app). I also checked the connection with scope and the uart signal from the header comes to the esp32 module pin so the connection is ok

hyperglitch, 2022-08-13:
The issue seems to be in the RogueMaster firmware. I had the same issue and after switching back to the official firmware it was fixed.

Emeric-hub, 2022-08-14:
Yep, maybe i'm wrong , but same kind of issue there ...

Latest RogueMaster (companion apps) - 0.64.2-0813-RMv2
Latest OTA / Firmware - esp32_marauder_v0_9_13_20220806_flipper

The ESP / Arduino Part complation/upload is OK (upload and webserver starting Nice)
OTA is going OK ... (Wifi devboard plugged to flipper)
FW Upload trough OTA work fine ... 
(maybe a few improvment is possible for the password ... than can be bypassed with the good URL Auth stored ?)

After OTA, the WIFI DEv card reboot and Marauder FW is Ok as the app ...

Reset of the card (or simply unplug/plug the card) ...
Then Wifi dev board is back to the OTA Webserver ...

I tried a few day ago with previous FW (from both RM and Marauder ... )

Seem that i'm not the only one ... (Flipper Discord)

justcallmekoko, 2022-08-16:
@Emeric-hub
When you first navigate to the MarauderOTA web page, make sure you immediately refresh the page before uploading the Marauder firmware. Once the firmware update is complete and it reaches 100%, push the physical reset button on the dev board. Let me know if this solves the issue with the firmware reverting back to MarauderOTA. If so, I am going to add this to the wiki. I am unable to reproduce this error so I cannot test my solution

Emeric-hub, 2022-08-19:
Sorry for the delay.

I already  tried what you asked me to ...

First try : plugged and powered trough flipper
-> OTA Ok
-> Upload Ok
-> Fw flashing Ok
-> the led blink (3 colors)

-> Wifi dev board trough marauder app OK
-> Reset Card -> marauder App KO (OTA OK)

Second try : plugged trough USB Port
-> OTA Ok
-> Upload Ok
-> Fw flashing Ok
-> the led blink (3 colors)
-> Reset Card

Plug to flipper -> Marauder app Ko ...

The board is a V1 ... (don't know if it helps)

Emeric-hub, 2022-08-19:
I also erase the board ... trough esp tool ...
It seems on the discord, that i'm not alone in this situation ...

Emeric-hub, 2022-08-19:
With Serial monitoring activated :

-> OTA 

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa70
load:0x40050000,len:0x292c
SHA-256 comparison failed:
Calculated: 8dce998efc3eb6dc43f3799a489dbcfbd5ada9828beadf71017c05c4a8d32bc3
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4004c18c

Marauder OTA Update

Configuring update server...
Initializing WiFi...
Starting softAP...

Displaying settings to TFT...
IP address: 192.168.4.1
Setting server behavior...
Finished setting server behavior
Starting server...
Completed update server setup

---------------------------------

SSID: MarauderOTA
Network Password: justcallmekoko
Username: admin
Password: admin
**Update server: Client connected -> 1
Update: esp32_marauder_v0_9_12_20220801_flipper.bin
Update Success: 941328
Rebooting...**
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400269a9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa70
load:0x40050000,len:0x292c
SHA-256 comparison failed:
Calculated: 8dce998efc3eb6dc43f3799a489dbcfbd5ada9828beadf71017c05c4a8d32bc3
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4004c18c
ESP-IDF version is: v4.4-dev-2313-gc69f0ec32
Failed to mount SD Card
SD Card NOT Supported
CLI Ready

              @@@@@@                        
              @@@@@@@@                      
              @@@@@@@@@@@                   
             @@@@@@  @@@@@@                 
          @@@@@@@      @@@@@@@              
        @@@@@@            @@@@@@            
     @@@@@@@                @@@@@@@         
   @@@@@@                      @@@@@@       
@@@@@@@              @@@@@@@@@@@@@@@@       
@@@@@                 @@@@@@@@@@@@@@@       
@@@@@                   @@@@@@@             
@@@@@                      @@@@@@           
@@@@@@                       @@@@@@@        
  @@@@@@                        @@@@@@@@@@@@
    @@@@@@@                          @@@@@@ 
       @@@@@@                     @@@@@@    
         @@@@@@@                @@@@@@      
            @@@@@@           @@@@@@         
              @@@@@@@      @@@@@@           
                 @@@@@@ @@@@@@              
                   @@@@@@@@@                
                      @@@@@@                
                        @@@@                

--------------------------------

         ESP32 Marauder      

            v0.9.12

       By: justcallmekoko

--------------------------------

----> This is OK

Reset is giving it back to OTA ...

> ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa70
load:0x40050000,len:0x292c
SHA-256 comparison failed:
Calculated: 8dce998efc3eb6dc43f3799a489dbcfbd5ada9828beadf71017c05c4a8d32bc3
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4004c18c

Marauder OTA Update

Configuring update server...
Initializing WiFi...
Starting softAP...

Displaying settings to TFT...
IP address: 192.168.4.1
Setting server behavior...
Finished setting server behavior
Starting server...
Completed update server setup

---------------------------------

SSID: MarauderOTA
Network Password: justcallmekoko
Username: admin
Password: admin

---> I don't have any SD card ... don't know if it helps...

Emeric-hub, 2022-08-19:
ESP32 boards librairies are now at v2.0.4
use the following options to compile :
[image]

Emeric-hub, 2022-08-19:
any clues here ?? :
https://docs.espressif.com/projects/esp-idf/en/v4.2.3/esp32/api-reference/system/ota.html

I will try to compile it myself next days ...

Emeric-hub, 2022-08-31:
new flasher script is working for me :
https://github.com/UberGuidoZ/Flipper/tree/main/Wifi_DevBoard/FZ_Marauder_Flasher

justcallmekoko, 2022-12-23:
@rwegrzyniak Any updates on this issue? Have you tried the flasher script?