justcallmekoko/ESP32Marauder #143: this project with new routers is useless attacks do not work now and obsolete

zagarolo, 2022-08-18:


zagarolo, 2022-08-18:
not working in 5 ghz

6q6, 2022-08-18:
> not working in 5 ghz 

Esp32 Isn't compatible with 802.11 AC. 
And new routers that send out 802.11 b/g/n 9/10x supply management frames which makes deauthentication ineffective in the first place.

justcallmekoko, 2022-08-18:
This issue is the equivalent of complaining a Tesla doesn't fly. This is a physical limitation of the ESP32 and no amount of firmware upgrades will fix this.

closed 2022-08-18 by justcallmekoko

zagarolo, 2022-08-18:
tell me what is it for then? hardware limitation we change hardware

zagarolo, 2022-08-18:
all routers now and have been in ac for years

justcallmekoko, 2022-08-18:
> 

If you change the hardware, you change the entire project since there isn't an arduino compatible MCU (that I know if) that is capable of 5ghz. So change the hardware sure but it would involve starting a completely new project, not modifying this one