justcallmekoko/ESP32Marauder #144: Discord link doesn't work

scriptguru, 2022-08-20:
I wanted to ask a question over Discord, but the link doesn't work (404).
So asking right here - would you be interested in getting rid of jQuery?
I could help with it. jQuery is not really needed here, it's used for a few things which most likely can be easily done without it.

justcallmekoko, 2022-08-20:
Which link did you use for the discord invite?
If you have a way of accomplishing the same thing the jquery does, please feel free to submit a PR

scriptguru, 2022-08-26:
Wiki Home page
"If you would like to chat with me and the rest of the community directly,
head over to my [discord](https://discord.gg/invite/w5JmasxvKA)."

justcallmekoko, 2022-12-23:
New invite link posted in wiki

closed 2022-12-23 by justcallmekoko