justcallmekoko/ESP32Marauder #145: Possibility to flash to a Wemos LOLIN32 board

augisbud, 2022-08-22:
I have tried out the method to use the provided OTA updater to flash a Marauder firmware, to be specific the _flipper.bin but every single one was the same.

The flash completes at a 100% however I get a message in the Serial console:
```
Update: esp32_marauder_v0_9_13_20220806_flipper.bin
E (75435) boot_comm: mismatch chip ID, expected 0, found 2
Could Not Activate The Firmware
```

The module I am using is a Wemos LOLIN32 v.1.0.0 4MB Flash
My partition scheme is Minimal SPIFFS (Large APPS with OTA)

If I were to flash the .bin file directly to the ESP32 which binary address should I use?

justcallmekoko, 2022-08-23:
What version of the arduino-esp do you have installed?

augisbud, 2022-08-30:
Sorry for the late reply, it's 2.0.0-rc1

justcallmekoko, 2022-12-23:
Use the standard _old_hardware.bin file. The flipper.bin is built for the ESP32-S2 module. You can also try to build this from source.

closed 2022-12-23 by justcallmekoko