justcallmekoko/ESP32Marauder #147: PMKID-Sniff on Flipper UART does not print EAPOL

angrybobcat, 2022-08-31:
Running `sniffpmkid` (with or without channel) only repeatedly prints  `Received EAPOL:` without actually outputting any related information about the captured EAPOL-Packet.

Since a SD-Card has to be manually soldered via breakout, a console-output would be nice to at least be able to copy the data manually.

Example output (same for Flipper display or on PC via UART-Bridge):
`
Starting PMKID sniff on channel 4. Stop with stopscan
Received EAPOL:
Received EAPOL:
Received EAPOL:
Received EAPOL:
Received EAPOL:
Received EAPOL:
`

justcallmekoko, 2022-08-31:
This functionality was removed from Marauder because it caused the sniffer to fail to receive subsequent frames after receiving the initial handshake. I would recommend adding the SD card modification since the main objective of Marauder is for packet capture

closed 2022-08-31 by justcallmekoko

angrybobcat, 2022-08-31:
Understood, thanks for the information.
Is there any possible known workaround to make use of that function without the SD card mod?

phra, 2022-09-14:
@justcallmekoko i have a couple of questions about this:

1. do you think it's possible to maintain this feature behind a configurable setting, so that it can be enabled if needed on the flipper zero?
2. do you think buffering before printing the packet information can solve the packet loss issue, eg. cache in memory the full handshake before printing it to screen?

thanks!

justcallmekoko, 2022-09-14:
1. It is likely possible to implement this as a setting
2. The issue is that printing prevents it from receiving any more packets. It would be easy if it was just missing packets while it was printing but it just messing with the wifi interface somehow. I troubleshot this on stream for a couple of hours with feedback.

mashkovtsevlx, 2022-12-29:
@justcallmekoko Can we add this information to [Wiki](https://github.com/justcallmekoko/ESP32Marauder/wiki/sniffpmkid)?

> Unlike other sniffing functions, the raw frame data is displayed on screen to be copied and saved elsewhere [This functionality was removed from Marauder because it caused the sniffer to fail to receive subsequent frames after receiving the initial handshake.

ComandanteBL, 2023-02-06:
> @justcallmekoko Can we add this information to [Wiki](https://github.com/justcallmekoko/ESP32Marauder/wiki/sniffpmkid)?

Are there a hardware limitations for implementing this, or what is the exact problem does anyone knows? I would like to take a look on this problem if anyone can provide more information about the problem?