justcallmekoko/ESP32Marauder #148: Flipper ESP32 - terminal in phone does not work, ESP32 LED flashes blue yellow red

igorzhilin, 2022-09-04:
Hi Koko,

1. I flashed the ESP32 firmware with flasher script using [this tutorial](https://github.com/justcallmekoko/ESP32Marauder/wiki/flipper-zero). The flasher did not show any errors.
2. When I plug the ESP32 in Flipper, its LED flashes blue, yellow, red and then turns off. 
3. When I connect the Flipper to phone in USB-UART bridge mode, port 0, with a USB cable and start the Serial USB Terminal from [your video](https://www.youtube.com/watch?v=_YLTpNo5xa0), the terminal shows "Connected to CDC device". But when I send commands, like `help`, nothing happens. When I send some random texts, the terminal responds in binary.
[image]

I also don't see Marauder WIFI in networks.

Any idea what I am doing wrong?

phr34k0, 2022-10-04:
Same here with putty on PC. The tutorial is damn vague be it Wiki or Youtube. All I'm getting is 

`>: help
Commands we have:
!                             ir
?                             led
bt                            loader
crypto                        log
date                          nfc
debug                         onewire
device_info                   power
factory_reset                 ps
free                          rfid
free_blocks                   start_rpc_session
gpio                          storage
help                          subghz
i2c                           update
ikey                          vibro
input`

itapai, 2022-10-21:
Having the same issue. Flashed firmware with https://github.com/SkeletonMan03/FZEasyMarauderFlash

justcallmekoko, 2022-12-23:
@igorzhilin Make sure the flipper baudrate is set to 115200 as well as on the phone, and pc. Once connected, try hitting the reset button on the dev board to see if the Marauder title screen shows up