justcallmekoko/ESP32Marauder #149: Possibility to build sources and install on Heltec Esp32 LoRa (v1-v2)

AeonDave, 2022-09-07:
Hello, i would like to know if is possible to use an Heltec esp32 lora board for this project.

https://doc.riot-os.org/group__boards__esp32__heltec-lora32-v2.html

I tried to build this but (with all libs) i find some blocking errors like:

```
framework-arduinoespressif32/tools/sdk/esp32/include/esp_wifi/include/esp_wifi_types.h:26:19: error: 'ESP_IF_WIFI_STA' was not declared in this scope
     WIFI_IF_STA = ESP_IF_WIFI_STA,
```
or
```

framework-arduinoespressif32/libraries/ESPmDNS/src/ESPmDNS.h:87:26: error: 'esp_interface_t' has not been declared
   void enableWorkstation(esp_interface_t interface=ESP_IF_WIFI_STA);
```

maybe i'm wrong or hardware is not compatible.

thank you

justcallmekoko, 2022-09-07:
What version of the IDE are you using? What version of ESP-IDF/arduino-esp32 are you using

AeonDave, 2022-09-07:
I'm using vscode with platformIO (last version)

https://docs.platformio.org/en/latest/boards/espressif32/heltec_wifi_lora_32.html#debugging

with this specs and libs

```
platform = espressif32
board = heltec_wifi_lora_32
framework = arduino
lib_deps = 
	ivanseidel/LinkedList@0.0.0-alpha+sha.dac3874d28
	bblanchon/ArduinoJson@^6.19.4
	adafruit/Adafruit NeoPixel@^1.10.5
	lvgl/lv_arduino@^3.0.1
	bodmer/TFT_eSPI@^2.4.75
	bodmer/JPEGDecoder@^1.8.1
	h2zero/NimBLE-Arduino@^1.4.0
	justcallmekoko/SwitchLib@^1.1.1
```