justcallmekoko/ESP32Marauder #152: Flipper Zero dev. board microsd card modification issue.

zano97, 2022-09-13:
Hi, 
i have made the microsd card modification to the developer board, from the android interface everything seems to work well, but when i look on the microsd there are no .pcap files saved (the microsd card seem empty). Looking at the properties of the microsd card i have noticed that some space is occupied on the card even if there are no files. 
Can you help me with this problem?
Thank you

justcallmekoko, 2022-09-14:
What are the specs of the SD card?
What brand is the SD card?
How have you formatted the SD card?
What Marauder firmware version are you running?
What scans did you try when you were expecting PCAPs to show up?
Do you have the `SavePCAP` setting enabled?

zano97, 2022-09-14:
The card is a MMAGR02GUECA-MB Samsung TransFlash 2GB microSD Flash Memory Card. 
I have tried to format the card both in FAT and FAT32 but gives the same problem.
The firmware version is 0.9.15.
Yes i have enabled SavePCAP, when mounted the system seems to recognize folders on the microsd but when i plug the microsd into the pc it's empty.

justcallmekoko, 2022-09-14:
is there a SCRIPTS directory on the SD card?
What scans have you tried?
When you start the Marauder, do it say "Failed to initialize SD card" in the serial output?

zano97, 2022-09-14:
no there is not any directory in the sd card.
I have tried scanap, then also a sniffpmkid.
no id does not say failed to initialize the sd card.
Thanks for your help!

justcallmekoko, 2022-09-14:
If there is no script directory, it sounds like the SD card is not mounting to the flipper dev board. Please send me the serial output from the time the dev board boots to when it presets the CLI prompt

zano97, 2022-09-14:
Here it is:

20:17:11.534 Connected to CDC device
20:17:15.037 ESP-ROM:esp32s2-rc4-20191025
20:17:15.040 Build:Oct 25 2019
20:17:15.041 rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
20:17:15.046 SPIWP:0xee
20:17:15.046 mode:DIO, clock div:1
20:17:15.048 load:0x3ffe6100,len:0x498
20:17:15.051 load:0x4004c000,len:0xa74
20:17:15.053 load:0x40050000,len:0x25ec
20:17:15.056 entry 0x4004c194
20:17:15.270 ESP-IDF version is: v4.4-dev-2313-gc69f0ec32
20:17:17.717 CLI Ready
20:17:17.719
20:17:17.719               @@@@@@                        
20:17:17.721               @@@@@@@@                      
20:17:17.726               @@@@@@@@@@@                   
20:17:17.730              @@@@@@  @@@@@@                 
20:17:17.736           @@@@@@@      @@@@@@@              
20:17:17.738         @@@@@@            @@@@@@            
20:17:17.741      @@@@@@@                @@@@@@@         
20:17:17.745    @@@@@@                      @@@@@@       
20:17:17.753 @@@@@@@              @@@@@@@@@@@@@@@@       
20:17:17.754 @@@@@                 @@@@@@@@@@@@@@@       
20:17:17.758 @@@@@                   @@@@@@@             
20:17:17.762 @@@@@                      @@@@@@           
20:17:17.765 @@@@@@                       @@@@@@@        
20:17:17.769   @@@@@@                        @@@@@@@@@@@@
20:17:17.774     @@@@@@@                          @@@@@@
20:17:17.778        @@@@@@                     @@@@@@    
20:17:17.782          @@@@@@@                @@@@@@      
20:17:17.786             @@@@@@           @@@@@@         
20:17:17.789               @@@@@@@      @@@@@@           
20:17:17.793                  @@@@@@ @@@@@@              
20:17:17.797                    @@@@@@@@@                
20:17:17.803                       @@@@@@                
20:17:17.805                         @@@@                
20:17:17.810
20:17:17.810
20:17:17.811
20:17:17.811
20:17:17.811 --------------------------------
20:17:17.813
20:17:17.813          ESP32 Marauder      
20:17:17.815
20:17:17.815             v0.9.15
20:17:17.817
20:17:17.817        By: justcallmekoko
20:17:17.820
20:17:17.820 --------------------------------
20:17:17.824
20:17:17.824
20:17:17.824 >

justcallmekoko, 2022-09-16:
Try sniffbeacon and see if there are still no files

zano97, 2022-09-16:
yes also with sniffbeacon the sd card is still empty

drontgtn, 2022-11-23:
Try to format SD card in Flipper first, worked for me.

frankiejoetabarrino, 2022-12-23:
PLEASE hel me.. i ve same problem.. my microsd seems not initialize

decayofmind, 2023-02-23:
I've just bought 32Gb SanDisk Ultra (red one) card and even formatted via Flipper (should not matter in fact) it's not supported.
Older 16Gb SanDisk Ultra (white one) right from my Flipper is supported though.